nasa/astrobee #2: Not able to install required dependencies

sahibdhanjal, 2018-05-05:
Hi,
I am getting stuck in the process of installing the dependencies, specifically at this step:

<code>./install_desktop_16_04_packages.sh</code>

in the README file [here](https://github.com/nasa/astrobee/blob/master/INSTALL.md).

The errors I am getting are as follows:

[image]

I've tried multiple times, however, none of my attempts have been fruitful. Because the dependencies are not satisfied, the code doesn't build as well. Could you please help me resolve this issue? 

The system specs are as follows:
Ubuntu 16.04.4 | 16 GB RAM | Core i9

ekpneo, 2018-05-05:
Hello:

I'm sorry that this isn't working smoothly for you. From the screenshot you provided, it looks like something in the `build_install_debians.sh` script either failed or it was not run. Please ensure you have run that script, as it builds and installs `libdbow`, `libavlar`, and a few others.

Don't hesitate to leave another comment if it continues not to work.

sahibdhanjal, 2018-05-05:
I have tried running the script multiple times and it seems to me that everything installed correctly using it. I have attached an install log file for <code>./build_install_debians.sh</code> [here](https://github.com/sahibdhanjal/astrobee/blob/master/Install%20Logs/installDeb.txt) . Hope the log file helps you in understanding the issue.

ekpneo, 2018-05-08:
Hello:

Again, I apologize that this has not gone as smoothly as it should. Thank you for providing the detailed log file. After analyzing it, we determined there was a bug in `build_install_debians.sh` that caused the built packages not to be installed. We have just pushed a fix for this. Please pull from this repository and try running that script again.

Don't hesitate to leave another comment if it still doesn't work.

sahibdhanjal, 2018-05-08:
Hi,
Appreciate your active support, the installation now happens correctly. However, I'm facing an issue while building the code as per:
<code>pushd $BUILD_PATH</code>
<code>make -j6</code>

<strong>The compilation stops giving me the following error : </strong>
<code>
[ 78%] Building CXX object hardware/i2c/CMakeFiles/i2c.dir/src/i2c_new.cc.o
hardware/i2c/CMakeFiles/i2c.dir/build.make:62: recipe for target 'hardware/i2c/CMakeFiles/i2c.dir/src/i2c_new.cc.o' failed
CMakeFiles/Makefile2:13666: recipe for target 'hardware/i2c/CMakeFiles/i2c.dir/all' failed </code>

<strong>Log files for <code>make</code> and <code>make -j6</code> can be found [here](https://github.com/sahibdhanjal/astrobee/blob/master/Install%20Logs/builde.txt) and [here](https://github.com/sahibdhanjal/astrobee/blob/master/Install%20Logs/buildError.txt) respectively.</strong>

ekpneo, 2018-05-08:
Hello:

It seems we're missing some output to see what the error actually is. I expect `g++` is printing something to `stderr` that's not getting caught in the logs. Can you try the following:

~~~sh
VERBOSE=1 make 2>&1 >log
~~~

Hopefully this will catch the compiler output and we can diagnose what's going on easier.

sahibdhanjal, 2018-05-08:
You can find the log file for the command you mentioned [here](https://github.com/sahibdhanjal/astrobee/blob/master/Install%20Logs/log). Hope it helps

sahibdhanjal, 2018-05-15:
Hi @ekpneo ,
Any update on the same?

bcoltin, 2018-05-15:
Hi Sahib,

We have tried on multiple computers and haven't been able to reproduce this issue. The error log doesn't give any error message from the compiler, so we aren't sure what is happening. For now, I suggest you try setting "USE_DRIVERS" to OFF in cmake (you can set this by editing your CMakeCache.txt in the build folder) which will disable building the drivers, including I^2C. These are not used in the simulation anyway. Sorry we cannot figure out anything more...

sahibdhanjal, 2018-05-15:
I switched off the USE_DRIVERS and it finally worked. Thanks for the help

closed 2018-05-15 by sahibdhanjal

sahibdhanjal, 2018-10-22:
Hi Yuriy,
If you go into the install scripts and check, they are specifically written for Ubuntu 16. To make it work for Ubuntu 14, you'll need to resolve further dependencies. Even I had started out with it, but setting it up was taking a lot of time, hence moved onto 16. I think you can wait for the contributors to reply to this thread, otherwise opening another ticket would be helpful.

ChernyshovYuriy, 2018-10-22:
Hi,

Thanks for replay.
I think it has more sense to switch to new Ubuntu rather then spend time to
resolve old version.
By any chance, do you know whether Ubuntu 18 supported? Because if do
upgrades it is better to upgrade to the latest version.

                                - Chernyshov Yuriy -
Senior Embedded Software Engineer at General Motors Canada

kbrowne15, 2018-10-22:
Hi,

I'm sorry we did not make this clear but the libraries that you are having problems with are internal NASA packages that you don't need to install.

ChernyshovYuriy, 2018-10-22:
That was my assumption :-)
However, "make" doesn't work for me. I can attach logs once I am at home.
It complains on gtest folder and once I comment out gtest compilation, it complain on NOTFOUND version of OpenCV libraries even thought I do have OpenCV installed and use it in my c++ projects.

ChernyshovYuriy, 2018-10-23:
I have to appologiese, I am on Ubuntu 16.04 LTS :-)
So, follow my previous post, when I ran make I faced error with gtest_noros. Its build folder is empty. However, if I comment out its build, I found that OpenCV version is not detected.
opencv_<sublib_name>-NOTFOUND is in all make files.
Is this something you can help to investigate?
Count in  mind I do have OpenCV vesrion 4 already installed.
[InstallProcess-2.txt](https://github.com/nasa/astrobee/files/2503833/InstallProcess-2.txt)

ChernyshovYuriy, 2018-10-23:
Hi Ted,
Thanks for analysis.
Yes, that was my assumptions as well, I was hope that you can provide more precise spot where this happening, I mean where exactly decision  been make which environment dependency pick up: the one provided by script or the one exists already in computer.
Anywhere, do you want me to go on with separate ticket or reopen this one or is this something you guys won't to handle?
Keeping conversation in closed ticket is not right :-)
What would be your choice?
Thanks a lot again.

ekpneo, 2018-10-23:
Hi @ChernyshovYuriy:

Yes, we should probably open another ticket as you are experiencing different errors than simply installing the required packages.

I notice that you are using a custom version of CMake (3.10). You may want to also try using the 3.5 version installed by Ubuntu, just to make your environment more similar to our expected setup.

CMake has an internal list of locations it will look in for libraries. We append ROS's installation location to this list within our `CMakeLists.txt` in the freeflyer source. You may want to look at [`CMAKE_IGNORE_PATH`](https://cmake.org/cmake/help/v3.5/variable/CMAKE_IGNORE_PATH.html) to see how to have it ignore everything under `/usr/local` and only use the system-installed packages. (Note that this is a list of directories, not prefixes, so it may get rather long.)

ChernyshovYuriy, 2018-10-23:
Hi Ted,
Sure, will do that once I am back home later on.
Thanks

LimSM, 2019-06-05:
Hello, I try to install it, I am also getting stuck in the same step:

./install_desktop_16_04_packages.sh

can you check the my log?
[log.txt](https://github.com/nasa/astrobee/files/3255787/log.txt)