nasa/astrobee #14: bash: pushd: no other directory

Sol-Rover, 2019-12-23:
Hey everyone! I'm having trouble with installing the dependencies. In the instructions, it said that I need to have root access. I have root access. Now, however:

What I type-
pushd $SOURCE_PATH

What I get back-
bash: pushd: no other directory 

Can someone help me please? I'm not sure what's going on or what I'm doing wrong. Thank you in advance for your help and response!

ekpneo, 2019-12-23:
In the [Machine Setup](https://github.com/nasa/astrobee/blob/master/INSTALL.md#machine-setup) section for Non-NASA users, it instructs you to do the following:

```bash
export SOURCE_PATH=$HOME/freeflyer
```

Note that the variable must be exported in *every* new terminal window you launch. If not, then the `pushd` command will fail, as `$SOURCE_PATH` is not set.

Sol-Rover, 2019-12-23:
Thank you very much. I did export it originally but that was when I didn't
have root access. I wasn't aware that I had to export in every new
terminal. Thank you! :)

Sol-Rover, 2019-12-23:
Do I also have to export ANDROID_PATH ="${SOURCE_PATH}_android again as well before I attempt to install the dependencies? Also, do I have to clone the repositories again or no since I've already done that?

ekpneo, 2019-12-23:
No, not for building/installing the dependencies.

As the instructions state, you only need to export `ANDROID_PATH` if you want to develop guest science applications, and the export is only used once (in these instructions, anyway) to checkout the `astrobee_android` repository. Other instructions will detail how to work with the Android stuff.

closed 2020-02-13 by bcoltin