nasa/astrobee #15: cannot display image in cv_bridge

lp02781, 2020-01-05:
hello, i'm new in ROS, and i get trouble that i don't know how to fix it
after i had done with my astrobee instalation, i make made another workspace to control astrobee position based on camera. but when i try to get topic for image, this result some error. 

`[ERROR] [1578215988.986783663, 37.688000000] : (ros.roscpp) Client [/pose] wants topic /hw/cam_dock to have datatype/md5sum [sensor_msgs/CompressedImage/8f7a12909da2c9d3332d540a0977563f], but our version has [sensor_msgs/Image/060021388200f6f0f447d0fcd9c64743]. Dropping connection.
`

lp02781, 2020-01-05:
i already fix it with change 
`sensor_msgs::CompressedImageConstPtr` with `sensor_msgs::ImageConstPtr`

but i still get error
`OpenCV Error: Assertion failed (size.width>0 && size.height>0) in imshow, file /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/highgui/src/window.cpp, line 331
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.3.1/modules/highgui/src/window.cpp:331: error: (-215) size.width>0 && size.height>0 in function imshow

Aborted (core dumped)
`

so far i know, this error mean my imshow doesn't have picture in there

bcoltin, 2020-02-13:
That's what it looks like, check the size of the image you are displaying. It sounds like this is an issue with your code and not with Astrobee.

closed 2020-02-13 by bcoltin