nasa/astrobee #29: Project build error

viraj96, 2020-07-29:
Hi, 

I have been able to successfully install the dependencies of the project. I was even able to configure the build of the project. However, when I try to compile it, I get the following errors,

```
Scanning dependencies of target gtest_noros
Scanning dependencies of target interest_point
[  1%] Creating directories for 'gtest_noros'
[  1%] No download step for 'gtest_noros'
[  1%] No patch step for 'gtest_noros'
[  1%] No update step for 'gtest_noros'
[  1%] Performing configure step for 'gtest_noros'
[  1%] Building CXX object localization/interest_point/CMakeFiles/interest_point.dir/src/agast_score.cc.o
-- Configuring done
CMake Error: Cannot open file for write: /usr/src/gtest/CMakeCache.txt.tmp
CMake Error: : System Error: Permission denied
CMake Error: Unable to open cache file for save. /usr/src/gtest/CMakeCache.txt
CMake Error: : System Error: Permission denied
external/gtest/CMakeFiles/gtest_noros.dir/build.make:123: recipe for target 'external/gtest/src/gtest_noros-stamp/gtest_noros-configure' failed
make[2]: *** [external/gtest/src/gtest_noros-stamp/gtest_noros-configure] Error 1
CMakeFiles/Makefile2:4405: recipe for target 'external/gtest/CMakeFiles/gtest_noros.dir/all' failed
make[1]: *** [external/gtest/CMakeFiles/gtest_noros.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  1%] Building CXX object localization/interest_point/CMakeFiles/interest_point.dir/src/brisk.cc.o
[  1%] Building CXX object localization/interest_point/CMakeFiles/interest_point.dir/src/essential.cc.o
make[2]: *** No rule to make target 'opencv_calib3d-NOTFOUND', needed by 'devel/lib/libinterest_point.so'.  Stop.
make[2]: *** Waiting for unfinished jobs....
Makefile:157: recipe for target 'all' failed
make: *** [all] Error 2
```
Can I get some help on how to fix this issue? Seems like the problem stems from `sudo` access and a weird `NOTFOUND` flag. I have `cmake 3.17` which meets the minimum requirement of the project. I even tried to run the make command with `sudo` but still faced the same `NOTFOUND` issue.

```
[  0%] Performing configure step for 'gtest_noros'
make[2]: *** No rule to make target 'opencv_calib3d-NOTFOUND', needed by 'devel/lib/libinterest_point.so'.  Stop.
CMakeFiles/Makefile2:15385: recipe for target 'localization/interest_point/CMakeFiles/interest_point.dir/all' failed
make[1]: *** [localization/interest_point/CMakeFiles/interest_point.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/src/gtest
[  0%] Performing build step for 'gtest_noros'
make[3]: *** No targets specified and no makefile found.  Stop.
external/gtest/CMakeFiles/gtest_noros.dir/build.make:128: recipe for target 'external/gtest/src/gtest_noros-stamp/gtest_noros-build' failed
make[2]: *** [external/gtest/src/gtest_noros-stamp/gtest_noros-build] Error 2
CMakeFiles/Makefile2:4405: recipe for target 'external/gtest/CMakeFiles/gtest_noros.dir/all' failed
make[1]: *** [external/gtest/CMakeFiles/gtest_noros.dir/all] Error 2
Makefile:157: recipe for target 'all' failed
make: *** [all] Error 2

```

Any help is appreciated!

bcoltin, 2020-07-29:
Hi Viraj,

My first guess could be, might you have another version of opencv installed? We use the opencv bundled with ROS and having another version installed can make the build process fail. 

I have not seen this write error before and don't have any idea there. You should not need to compile as root.

viraj96, 2020-07-29:
While running the configuration command I got the following output with regards to OpenCV,

`-- Found OpenCV: /opt/ros/kinetic (found suitable version "3.3.1", minimum required is "3") `

I have checked that `libopencv_calib3d3.so` exists inside `/opt/ros/kinetic/lib/x86_64-linux-gnu`. Now it seems like cmake should have looked for `opencv_calib3d3` instead of `opencv_calib3d`?

viraj96, 2020-07-29:
Additionally, I got a bunch of cmake warnings like the following for multiple packages,

```
CMake Warning (dev) at /usr/local/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (ceres) does
  not match the name of the calling package (Ceres).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/FindCeres.cmake:66 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:210 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

```

Although it seems unlikely but could this be causing any issues. This did not happen for OpenCV though.

closed 2021-03-10 by bcoltin

izzunnafis, 2021-04-23:
I got the same problem, is there any solution?

bcoltin, 2021-04-26:
We are doing a find_package(OpenCV 3 REQUIRED) which is causing this error, nothing complicated. So I suspect it is a problem with the ROS kinetic FindOpenCV cmake script. I have seen this error happen when multiple versions of opencv are installed. Unfortunately I don't have a workaround other than uninstalling all the other versions. If you have time to look at it, you might also be able to fix it by reordering CMAKE_PREFIX_PATH so the right one is found first.