nasa/astrobee #36: Build Documentation on Commit to Master Automatically

bcoltin, 2020-08-20:
Already done for DELTA, just need to change process of building: https://github.com/nasa/delta/blob/master/.github/workflows/docs.yaml

marinagmoreira, 2020-08-27:
Added the action for doxygen. I made it trigger in pull request to test it and failed with:

remote: Permission to nasa/astrobee.git denied to github-actions[bot].
fatal: unable to access 'https://github.com/nasa/astrobee/': The requested URL returned error: 403

So I'm assuming it will work once we merge it because that line (last one of the yaml) is the same as in delta

marinagmoreira, 2020-08-27:
I actually tested on my own fork and worked well!

bcoltin, 2020-08-27:
I think the issue is that the action is set to run on both push and on pull requests. We should set pushing to github to only run on push, otherwise either we will overwrite the documentation with unaccepted changes. On delta we have it set to only push on changes to master, because this is the version we expect most users will be on.

Btw, did you see my comments in the review on your pull request? I think when you do the force pushes it causes trouble for the reviewing tools, should avoid if possible...

marinagmoreira, 2020-08-27:
I put it commented in the action yaml file that # to remove after action is verified
I just put it on pull request to be able to see the action be executed and verify that it works properly. That way the changes in the documentation are in https://marinagmoreira.github.io/astrobee/html/index.html, which should make the review easier

bcoltin, 2020-08-27:
Oh ok! I got an email that a review was requested so I thought you were ready to merge :)

closed 2020-08-28 by marinagmoreira