nasa/astrobee #55: Likely bug in logic of multi-Astrobee sync via setTimeSync command

trey0, 2020-10-03:
https://github.com/nasa/astrobee/blob/bb0fc3e4110a14929bd4cf35c12b3c169bc6c756/mobility/choreographer/src/choreographer_nodelet.cc#L1123

I'm pretty sure the '+=' should be a '-=' to achieve the synchronization effect that we are looking for.

Btw, I would not suggest investing time in fixing this feature. Rather, I take this as an indication that it has never been tested, and likely has other bugs as well. Besides, the sync protocol is inherently not robust. Probably it should be abandoned.

bcoltin, 2020-10-05:
It looks like it is disabled by default. Do we know what this is even intended for? @marinagmoreira

asymingt, 2020-10-05:
If I recall correctly, this was a badly-implemented attempt to make
multi-astrobee actions time-synchronized in the presence of propagation
delay from GDS to the individual unit, when enable_immediate is set (which
rewrites the trajectory to start immediately). It essentially maps the
start time of the segment to the nearest whole second, under the assumption
that the clocks are synchronized with NTP in the background, and NTP is a
couple orders of magnitude better synchronization than 1 second. It's a
silly thing that is turned off by default, and has no effect on behavior.
The better way to achieve this functionality is completely remove
enable_imediate (this is what breaks synchronization) a favor GDS or the
teleop tool supplying a global initial trajectory timestamp for some point
in the future.

bcoltin, 2021-05-07:
We talked to the facilities team, and there is little interest in this. We will remove the executive command and add a comment explaining it is untested to choreographer.

kbrowne15, 2021-07-26:
This command has been removed.

closed 2021-07-26 by kbrowne15