nasa/astrobee #58: Confusingly unused flight modes in granite.config

trey0, 2020-10-03:
granite.config specifies some special-purpose flight modes: docking, undocking, perching, unperching.

Of these modes, it looks like only docking (MotionGoal::DOCKING) is used as you would expect in the relevant behavior. The others use MotionGoal::NOMINAL.

I'm not sure if the unused flight modes should be removed or the behaviors should be changed to use them as you would expect.

Mode definitions:
https://github.com/nasa/astrobee/blob/e9a3a6694176d4e07b192972115e1cf5429573f4/astrobee/config/worlds/granite.config#L241

Non-usage examples:
https://github.com/nasa/astrobee/blob/e9a3a6694176d4e07b192972115e1cf5429573f4/behaviors/perch/src/perch_nodelet.cc#L179
https://github.com/nasa/astrobee/blob/e9a3a6694176d4e07b192972115e1cf5429573f4/behaviors/perch/src/perch_nodelet.cc#L237
https://github.com/nasa/astrobee/blob/e9a3a6694176d4e07b192972115e1cf5429573f4/behaviors/dock/src/dock_nodelet.cc#L310

bcoltin, 2020-10-05:
I think we should remove them, it is just more opportunities to screw up the configuration. @marinagmoreira do you agree? Can you add this to the bottom of your todo list?

marinagmoreira, 2020-10-07:
I agree, I don't think it would be productive to even customize these flight modes

kbrowne15, 2020-10-07:
These flight modes are also in the iss world config, mgtf world config, and perch world config so they should be removed from those configs as well. Actually, I don't believe we even have a perch world anymore so that config should probably be deleted altogether.

closed 2020-10-27 by marinagmoreira