nasa/astrobee #81: Static assertion error during build

glegarda, 2020-11-22:
Hello,

I am trying to build the project, but when I run `make -j2` I get the following error:

```
Scanning dependencies of target marker_tracking
[ 44%] Building CXX object localization/marker_tracking/CMakeFiles/marker_tracking.dir/src/arconfigio.cc.o
In file included from /usr/include/c++/8/map:61,
                 from $SOURCE_PATH/localization/marker_tracking/include/marker_tracking/arxmlio.h:25,
                 from $SOURCE_PATH/localization/marker_tracking/include/marker_tracking/arconfigio.h:24,
                 from $SOURCE_PATH/localization/marker_tracking/src/arconfigio.cc:19:
/usr/include/c++/8/bits/stl_map.h: In instantiation of ‘class std::map<int, Eigen::Matrix<float, 4, 3>, std::less<int>, Eigen::aligned_allocator<std::pair<int, Eigen::Matrix<float, 4, 1> > > >’:
$SOURCE_PATH/localization/marker_tracking/src/arconfigio.cc:54:27:   required from here
/usr/include/c++/8/bits/stl_map.h:122:21: error: static assertion failed: std::map must have the same value_type as its allocator
       static_assert(is_same<typename _Alloc::value_type, value_type>::value,
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
localization/marker_tracking/CMakeFiles/marker_tracking.dir/build.make:62: recipe for target 'localization/marker_tracking/CMakeFiles/marker_tracking.dir/src/arconfigio.cc.o' failed
make[2]: *** [localization/marker_tracking/CMakeFiles/marker_tracking.dir/src/arconfigio.cc.o] Error 1
CMakeFiles/Makefile2:19279: recipe for target 'localization/marker_tracking/CMakeFiles/marker_tracking.dir/all' failed
make[1]: *** [localization/marker_tracking/CMakeFiles/marker_tracking.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 44%] Linking CXX executable ../../devel/lib/camera/test_camera_params
[ 44%] Built target test_camera_params
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```

I really do not know how to fix this. Could someone please help me? Any help would be appreciated!

algorithmhash, 2021-01-02:
I could take a look and get  back to you about it @glegarda

algorithmhash, 2021-01-03:
Also, a list of your environment and OS would help a bit. 🙂

glegarda, 2021-02-26:
My bad @algorithmhash, I was trying to do a native build on Ubuntu 18, which I then realised is not officially supported! I ended up using another machine with Ubuntu 16 and everything worked perfectly.

If you would like to find out more about the issue anyway, we could look into it. Otherwise, please feel free to close it.

closed 2021-04-28 by bcoltin