nasa/astrobee #127: Reimagine Attitude / Position Tolerances

bcoltin, 2021-02-12:
It is not working. Not sure what we should do instead. Let's think more about this.

bcoltin, 2021-03-26:
This was an issue on the last activity again. It usually happens when localization jumps, however, it should not need to disrupt control. We talked about expanding the tolerance to the width of Astrobee (and maybe doubling attitude tolerance).

marinagmoreira, 2021-04-01:
Just did a pull request, if this does not solve the issue, we should start thinking of fancier ways of doing this by using zones and triggering only when reaching the edge (in case the robot is not moving ff and therefore not using obstacle detection)

bcoltin, 2021-05-03:
Will look at bag from last activity, make sure it is caused by localization jumping. If so will talk with Ryan and consider alternatives.

marinagmoreira, 2021-08-12:
We'll create a new check in the end of the movement to confirm that the robot reached the desired end point. During the movement we'll increase the tolerance_timer (time duration which the fault is active before triggering the error) and keep the tolerance relatively high so as to avoid interruptions during localization jumps or difficult trajectories.

trey0, 2021-08-12:
I think #259 looks great!

Some big picture thoughts:

There were some documentation improvements in the PR (great!) but it's still a bit hard to decipher what the behavior is supposed to be. (For example: the semantics of tolerance_pos vs. tolerance_pos_setpoint.)

I can see three main reasons to raise a fault based on position tolerance:
  1. High position error is a sign of poor mobility subsystem health. - If this is the main issue, it makes sense to have the tolerance tunable based on the propulsion mode the way you have it (since we expect larger errors in some modes). But why is tolerance_pos_setpoint tighter than tolerance_pos? Do we really expect tighter position accuracy at the end?
  2. High position error creates a collision risk. - If this is the main issue, it might make sense to tie tolerance_pos to collision_distance from mapper.config. But maybe our stance should be that we already have explicit collision avoidance that can be smarter about what kinds of errors bring us too close to obstacles, so the simple position error checking isn't about collisions.
  3. High position error doesn't achieve the user's objectives. - If this is the main issue, it might make sense to allow user tuning of the tolerance, like the way collision_distance can be set by Settings.setOperatingLimits. Different users may have different precision requirements, and some users may care only about the endpoint of the motion vs. others may want accurate trajectory tracking throughout. But of course just setting a tighter tolerance doesn't by itself make the system fly more accurately, it just raises a fault when execution fails, so offering a tunable tolerance might tend to disappoint users more than help them.

I think the above discussion supports emphasizing motivation (1) and the approach you already took.

But there is still the question of why tolerance_pos_setpoint should be tighter than tolerance_pos. Do we really think accuracy will be better at the end, or are we setting ourselves up for frequent failure that way? Is there some idea that we are also trying to satisfy motivation (3) and we believe users care more about the endpoint than the trajectory? (True for ISAAC mapping, not for all Astrobee scenarios.) But if that's the case, I think the best place to start might be more about developing behaviors that specifically improve end-of-motion accuracy, rather than enforcing a tighter error check that might often fail.

marinagmoreira, 2021-08-13:
Let me add some more documentation on the choreographer side, I don't think these tolerances are documented explicitly.

1. What we've seen is that the cause of the mobility issues is mainly due to temporary poor localization which manifests itself more when the robot is in motion. From conversations with the facilities team, there is a high interest in ensuring that when the control returns success that the robot is indeed in the goal target position.
2. That was the rational for increasing the position tolerance without worrying too much about collisions. 
3. I don't think there is actually a problem with position error, but more with temporary perception error. This is mainly aimed at minimizing interrupting operations during activities in order to resend commands. The way things are setup right now, the user can choose the flight mode which will have different settings according to the present goal - nominal, difficult, precision, quiet (it's a way to give some control without having to think about each parameter individually)

At the end of the motion, after the robot stopped, we expect the position errors to be smaller than during motion. It seems that it is important for the majority of the activities to make sure the robot is at the correct position, such that the operator knows to resend the move command. That's why there is an emphasis in making sure that the end position is more accurately evaluated. I don't think that *after tuning*, this check will fail often, and when it does the operator knows to resend the command if the specific end position is of concern - this check doesn't actually interfere with the move command dynamics but only with its result.

trey0, 2021-08-16:
Thanks for the clarification!

From what you said:
- Your reason to improve position error checking is a combination of motivation (1) stop when health is degrading and motivation (3) achieve user accuracy objectives. We do not intend to address motivation (2) avoid obstacles using position error checking.
- While acknowledging that some users might care about both trajectory tracking accuracy and target position accuracy, we think the more common case is that they care more about target position accuracy, and also the facility team wants to prioritize that on behalf of users.
- Checking the position error at the end of motion also causes less disruption when the check fails because it doesn't interrupt motion. I would add a note of caution: that's a valid point if you are only worried about a single motion, but an end-of-motion fault does interrupt execution of the rest of the plan.

I think these are good arguments and it's a good approach! Some possible ideas for future improvements in this area:
- To the extent that the error checks are about motivation (3) achieving user accuracy objectives, we may want to provide users with a convenient way to tune these tolerances, but only within a valid range. They should not make the tolerance too tight (because motion is unlikely to achieve the tolerance and they will just get frustrated with errors) nor too loose (because it won't catch mobility health issues, motivation (1)). The point of tolerance tuning is to avoid unnecessary disruption due to unnecessary faults.
- If accurate target positions specifically are a priority, we may want to experiment with ways to improve end-of-motion accuracy. Example behaviors could include: slowing down a little bit early, continuing to home in on the target position after the pre-planned trajectory ends rather than failing, and briefly holding position at the target before declaring success to ensure that the localizer estimate is stable. But all of those behaviors potentially have a time cost, so it would make sense to provide users with a way to enable/disable them, along the lines of the existing precision mode.

bcoltin, 2021-08-17:
> Checking the position error at the end of motion also causes less disruption when the check fails because it doesn't interrupt motion. I would add a note of caution: that's a valid point if you are [...]

This is a good point. I would suggest the default be to prioritize not failing frequently. Then we could provide the facility / guest scientists with two other configurations --- one to inform you if you end in the wrong place, and another to fail if the trajectory is off some reasonable amount.

I also love the idea of a behavior to reduce being in the wrong place at the end. Let's make that a second pull request for that though and add one thing at a time.