nasa/astrobee #146: Localization Slightly Jittery while Station Keeping

bcoltin, 2021-03-10:
While station keeping, the robot would move back and forth approximately ten centimeters. However, it would not get lost. It would be good if we could smooth the pose estimate to reduce these oscillations.

trey0, 2021-03-19:
This issue relates to:
- ISS activity: Crew-Minimal 11
- Date: 2021/03/10
- FSW release: 0.14.1

rsoussan, 2021-03-26:
This issue is related to a bug in the visual odometry feature adding.  Features are added in the wrong order, and the latest image with vo features isn't always added, leading to larger extrapolation times in the IMU Augmentor.  More details are here: https://babelfish.arc.nasa.gov/confluence/display/astrosoft/CM11+Errors+And+Solutions.  A fix is in development

closed 2021-05-03 by bcoltin