nasa/astrobee #147: Investigate Localization Failures / Near Failures

bcoltin, 2021-03-10:
When we pitched up the first time, localization drifted a little but recovered. When we moved vertically towards the deck, the robot became briefly lost and had some difficulty recovering. We should investigate both the initial failures and why it didn't recover more quickly.

trey0, 2021-03-19:
This issue relates to:
- ISS activity: Crew-Minimal 11
- Date: 2021/03/10
- FSW release: 0.14.1

rsoussan, 2021-03-26:
The issue with visual odometry feature order affected both of these failures, with the pitch event additionally suffering from moving flags due to the robot fan and the deck issue suffering from a poor map measurement.  These are described in more detail here: https://babelfish.arc.nasa.gov/confluence/display/astrosoft/CM11+Errors+And+Solutions.  As mention in issue 146 https://github.com/nasa/astrobee/issues/146, a fix is in development.

bcoltin, 2021-05-03:
Currently working on separating into two graphs, one for velocities / bias tracking, one for pose.

bcoltin, 2021-07-26:
Lots of progress on making localization more robust. Still issues, especially with rotation.

closed 2021-11-01 by bcoltin