nasa/astrobee #156: Sci Cam Driver Stills Broken

bcoltin, 2021-03-26:
The sci cam driver has been regularly crashing. Our script to work around this also crashes, although less frequently.

oleg-alexandrov, 2021-03-27:
This is serious business. It was not crashing for me for a while. That the workaround script crashes too is even worse, as that means the issue is worse than the camera being overloaded after a while, it can't acquire even a single first picture.

I guess all the code must be carefully reviewed. I also pointed out to Katie the original code I used as a base for the sci cam script. Maybe that one does not crash and then I made a bug when adapting it.

Note also that the sci cam uses the Android camera2 api. In the history there is a version with the older camera api. Wonder if that one crashes too. The old api only can set the focal length at infinity. Not ideal, but may be worth looking as to where it starts to crash.

I have no more ideas. Would be curious to know what the issue turns out to be.

bcoltin, 2021-05-03:
Katie is planning to look into this further.

closed 2021-12-16 by kbrowne15