nasa/astrobee #158: Add pico proxy to the ISAAC launch

bcoltin, 2021-03-26:


oleg-alexandrov, 2021-03-27:
Actually this should even be in the astrobee repo as well. If the haz cam is meant to take pictures, it should as well decode properly the depth right away. We need to do that for ISS work, and I don't know if we will ever install isaac on the ISS. Having pico_proxy only for isaac looks of limited value to me.

bcoltin, 2021-03-29:
ISAAC is already installed on the ISS. I don't think the obstacle detection requires pico_proxy, so the default Astrobee software doesn't need it?

oleg-alexandrov, 2021-03-29:
I am not sure what the default astrobee software does with the depth camera.

marinagmoreira, 2021-05-03:
This was done in the latest ISAAC release for the ISAAC3 activity. This was tested in the lab and in space

closed 2021-05-03 by marinagmoreira