nasa/astrobee #162: Rewrite the merge bags tool in C++ because of rosjava error

marinagmoreira, 2021-04-01:
Because of rosjava, the merge bag tool with the python API doesn't work with the error:
genmsg.msg_loader.MsgNotFound: Cannot locate message [Header]: unknown package [std_msgs] on search path [{}]

This doesn't happen with the C++ API because we have the sci cam add tool that runs fine, and we're actually using it to preprocess the bags because the other tool doesn't work.

oleg-alexandrov, 2021-04-02:
To just fill in some detail, the isaac tool named isaac/src//dense_map/geometry_mapper/tools/add_sci_cam_to_bag.cc (which shows how to use the C++ API to process a bag) should be copied to the astrobee repo, renamed to merge_bags.cc, and some simple modifications need to be made to it to append multiple bags to a bag instead of appending sci cam images, and the old merge_bags.py tool this would replace should be wiped. 

This is overall little work and will greatly improve our bag processing toolchain. The new tool should be tested though with a problematic bag to ensure it works where the old one did not. Also the documentation in astrobee where this tool is referenced should be updated.

oleg-alexandrov, 2021-04-28:
I replaced merge_bags.py with a C++ tool called merge_bags. This fixes the rosjava issue. The API of the new tool is:

merge_bags -output_bag out.bag input1.bag input2.bag ...

closed 2021-04-29 by oleg-alexandrov