nasa/astrobee #178: Factor graph based vio

poudyalbot, 2021-05-18:
Dear sir the factor graph based localisation implemented in astrobee.
Where can I find the theoretical background on the implementation.
Thank you

rsoussan, 2021-05-19:
Hello, 
We have a paper that is currently under submission.  For more information in general about the factor graph implementation I'd recommend some of the GTSAM papers from the GTSAM docs (https://github.com/borglab/gtsam/tree/develop/doc) including documentation on IMU preintegration factors and visual odometry smart factors (https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=6907483&casa_token=0D5QL0tG5owAAAAA:9HydHD-64FnQRJKS9pIwbCkpFQxQJdZStHkXH7m7aZZRxqIendUn9jKbbLWVu1O9-p4R8OoZpROX&tag=1).  If you have more question please let me know.

poudyalbot, 2021-05-20:
Dear sir,
Thank you for your quick reply. Is it possible to use the factor graph
based vio for quadrotor type vehicle or is it only designed for astrobee
just like the ekf based vio.
Thank you

poudyalbot, 2021-05-20:
Where can i find the paper.

rsoussan, 2021-05-24:
It is possible to use for a quadrotor as the factor graph takes as input a gravity vector in the world frame, and this is simply set to zero when we use it on the ISS.  The sensor inputs are the same as the old EKF, and the output is the same as well.  I'll make sure to link the paper in our documentation once it is available online.

poudyalbot, 2021-05-25:
Thank you for reply.

marinagmoreira, 2021-06-02:
I'm going to close this issue for now, but feel free to open it back up if something comes along

closed 2021-06-02 by marinagmoreira