nasa/astrobee #186: Vive localisation calibration

aimatter, 2021-05-26:
I am running the provided melodic docker image.
For my use cases I try to use the localisation_vive package to have an ros package that is publishing an accurate static vive position. For this I followed the guideline and have now issues with the calibration part. The following questions raised up:
- how to collect a rosbag if I can not launch the node without the bag (what tool are you using to solve this issue)
- none of the published topics send messages as can be seen in the screenshot
- there is no config file in astrobee/localization/vive_localization.config

[image]

marinagmoreira, 2021-05-27:
Hello!

The vive localization package on astrobee is aimed at lab testing only (using the astrobee hardware) and not simulation.
- Our procedure to using the vive is to collect the lab data by powering on the connected vive sensors, and then running the flight sofware on the robot as:
roslaunch astrobee astrobee.launch mlp:=mlp llp:=llp extra:=vive
Only after the data is collected in a ros bag with the info published by the vive sensors we process the bag to obtain the ground truth pose output.
roslaunch vive_localization vive_localization.launch bag:=/path/to/test.bag rviz:=true
- It looks like gazebo is crashing on your side (the simulator). Without it the simulation the nodes do not run, explaining your output. There are many reasons this might be happening. For example, you can't run the docker images inside a vm as gazebo needs access to the gpu. Regardless, I would advise using the latest flight software version from this repo and building the docker images yourself with ./scripts/docker/build.sh -n (-n is to build the ubuntu 18 version). This will eliminate some failure points. However bear in mind that the simulation does not have the capability of simulating vive output and the above point's procedure is the only way.
- The config file is in the repo in https://github.com/nasa/astrobee/blob/master/astrobee/config/localization/vive_localization.config, and inside the docker image in /src/astrobee/astrobee/config/localization

It is worth noting that the vive_localization is made for our lab setup and without some tinkering with the code, it will be very unlikely that you can reuse the node (quantity of vive trackers and relative position to each other needs to be specified)

Cheers, let me know if you have more questions

marinagmoreira, 2021-06-02:
I'm going to close this issue for now since we didn't get a response, but feel free to open it back up if something comes along

closed 2021-06-02 by marinagmoreira