nasa/astrobee #264: Mention 20.04 Install in Documentation

bcoltin, 2021-08-17:
It looks like the install instructions currently only mention 18.04 and 16.04.

ana-GT, 2021-08-18:
Thanks for the prompt reply Brian! I gave it a shot and my main blocking issue seems to be Alvar. From what I can tell, this is what happens:

1. Ubuntu 20.04 comes with OpenCV 4.x, which is used by ROS noetic packages as well.
2. Astrobee needs OpenCV 3.x. It seems the main reason for this is Alvar, whose source code depends on legacy (e.g. IplImage), which stopped being supported on opencv versions 4 onwards.
3. When the Astrobee's configuration files try to build an OpenCV .deb (version 3.x), it fails in my machine because it says that it needs to install the dependency libvtk6-dev, however 20.04 has installed libvtk7-dev. I cannot uninstall libvtk7-dev since a lot of ros packages I use depend on that one.

My new question :) now is : Is Alvar the **only** reason for using OpenCV version 3? If so, then I could give a shot at patching **alvar** (the version astrobee uses), so it can be used with OpenCV 4 (and I think the Astrobee's localization package also would need some few updating changes, but it does not seem anything major).  Thanks!

marinagmoreira, 2021-08-18:
The 'Ubuntu 20.04 install' installs OpenCV 3.3.1 in /usr/local, such that you can have both versions simultaneously. The instructions for all versions are the same though given that the install_debians step detects the ubuntu version automatically.

ana-GT, 2021-08-18:
Hi Marina!

Thanks. Mm, wonder what is the issue with my system then. When I run ./build_install_debians.sh, this is what happens: 
```
ana@strelka:~/ros/astrobee/scripts/setup/debians$ ./build_install_debians.sh 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
devscripts is already the newest version (2.20.2ubuntu2).
equivs is already the newest version (2.2.0).
libproj-dev is already the newest version (6.3.1-1).
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libopencv3.3.1-build-deps : Depends: libvtk6-dev but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

My system has libvtk7-dev installed. If I try to install libvtk6-dev, that uninstalls libvtk7-dev, which I need since that is a dependency for packages like ros-noetic-pcl-ros.

marinagmoreira, 2021-08-18:
Humm, the install instructions assume a clean OS to begin with. Maybe you can try to replace 'libvtk6-dev' with 'libvtk6-dev | libvtl7-dev' in the control file to see what happens (scripts/setup/debians/opencv/control).

ana-GT, 2021-08-18:
I see, thanks! For reference, this is a  brand new machine, which I set up 5 days ago with 20.04 and Noetic. Before attempting to install Astrobee, I had only installed ros-noetic-desktop-full in this machine (plus some extra ROS packages, but nothing out of ordinary - as in, I didn't add any super new package or ppa that might have messed up my machine). 

As for your suggestion (using libvtk7-dev) it helped to make the build progress :). However, it stalled at some point, apparently for a bunch of reasons. The one that caught my eye was this one related to OpenCV:
```
/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: sys/videoio.h: No such file or directory
    1 | #include <sys/videoio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
```
From what I can see, it seems as if this is not built/installed because libv4l-dev is not being found properly by OpenCV 3.x. I see I have that package installed on my machine. I am guessing the OpenCV CMake is not finding it because it is searching for **linux/videodev.h**, but from what I read online, it seems that now that file is installed in **/usr/include/libv4l1-videodev.h** .  

I got hundreds of error/warning lines, so it is a bit hard to parse what is exactly failing (besides what I mentioned above). I copied a few last dozen lines of the terminal output of ./build_install_debians.sh for reference. I probably will come back to this at some point soon and will let you know if I figure something out. Thanks for the help!

```/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: sys/videoio.h: No such file or directory
    1 | #include <sys/videoio.h>
      |          ^~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [CMakeFiles/cmTC_195a6.dir/build.make:66: CMakeFiles/cmTC_195a6.dir/CheckIncludeFile.c.o] Error 1
make[3]: Leaving directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
make[2]: *** [Makefile:121: cmTC_195a6/fast] Error 2
make[2]: Leaving directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'

Determining if the function sgemm_ exists failed with the following output:
Change Dir: /home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make cmTC_96a9d/fast && make[2]: Entering directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
/usr/bin/make -f CMakeFiles/cmTC_96a9d.dir/build.make CMakeFiles/cmTC_96a9d.dir/build
make[3]: Entering directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_96a9d.dir/CheckFunctionExists.c.o
/usr/bin/cc   -g -O2 -fdebug-prefix-map=/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-narrowing -Wno-comment -Wno-implicit-fallthrough -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections  -msse -msse2 -msse3 -fvisibility=hidden -DCHECK_FUNCTION_EXISTS=sgemm_ -O3 -DNDEBUG   -o CMakeFiles/cmTC_96a9d.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.16/Modules/CheckFunctionExists.c
cc1: warning: ‘-Werror=’ argument ‘-Werror=non-virtual-dtor’ is not valid for C
Linking C executable cmTC_96a9d
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_96a9d.dir/link.txt --verbose=1
/usr/bin/cc -g -O2 -fdebug-prefix-map=/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2   -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-narrowing -Wno-comment -Wno-implicit-fallthrough -fdiagnostics-show-option -Wno-long-long -pthread -fomit-frame-pointer -ffunction-sections  -msse -msse2 -msse3 -fvisibility=hidden -DCHECK_FUNCTION_EXISTS=sgemm_ -O3 -DNDEBUG  -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now   -rdynamic CMakeFiles/cmTC_96a9d.dir/CheckFunctionExists.c.o  -o cmTC_96a9d 
/usr/bin/ld: CMakeFiles/cmTC_96a9d.dir/CheckFunctionExists.c.o: in function `main':
/usr/share/cmake-3.16/Modules/CheckFunctionExists.c:17: undefined reference to `sgemm_'
collect2: error: ld returned 1 exit status
make[3]: *** [CMakeFiles/cmTC_96a9d.dir/build.make:87: cmTC_96a9d] Error 1
make[3]: Leaving directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'
make[2]: *** [Makefile:121: cmTC_96a9d/fast] Error 2
make[2]: Leaving directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv/obj-x86_64-linux-gnu/CMakeFiles/CMakeTmp'

Determining if the function cheev_ exists failed with the following output:
[... cut ...] KE_SHARED_LINKER_FLAGS_RELEASE=-Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now" -DBUILD_SHARED_LIBS=ON -DBUILD_DOCS=ON .. returned exit code 1
make[1]: *** [debian/rules:34: override_dh_auto_configure] Error 25
make[1]: Leaving directory '/home/ana/ros/astrobee/scripts/setup/debians/libopencv/opencv'
make: *** [debian/rules:20: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1182:
dpkg-buildpackage -us -uc -ui failed
```

marinagmoreira, 2021-08-18:
Thanks for the info, I'll try to reproduce your setup then :)

marinagmoreira, 2021-08-19:
Our docker image does install ROS prior to building/installing the debians (https://github.com/nasa/astrobee/blob/develop/scripts/docker/astrobee_base.Dockerfile).

I'm going to close this issue because I just merged in a PR which mentions 20.04 in the documentation, maybe we could open a separate one to sort this out?

closed 2021-08-19 by marinagmoreira

ana-GT, 2021-09-07:
That is awesome! Thank you very much :)