nasa/astrobee #289: Eigen alignment

marinagmoreira, 2021-09-24:
Got some Eigen alignment issues when switching to catkin compiling in perch and mapper node.
For perch node I added:
behaviors/perch/src/perch_nodelet.cc
@@ -1031,6 +1031,11 @@ class PerchNodelet : public ff_util::FreeFlyerNodelet {
+ public:
+  // This fixes the Eigen aligment issue
+  // http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html
+  EIGEN_MAKE_ALIGNED_OPERATOR_NEW

For mapper node I added:
mobility/mapper/CMakeLists.txt
@@ -20,6 +20,7 @@ project(mapper)
+set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -D EIGEN_MAX_STATIC_ALIGN_BYTES=0")

These additions solved the problem, but it deserves further looking into.

------------------------------------------ error -----------------------------------------------------------

[llp-0]: ROS_MASTER_URI=http://llp:11311
nodelet: /home/mgouveia/ros_ws/arm_cross/rootfs/usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 4; int MatrixOrArrayOptions = 0]: Assertion `(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.

------------------------------------------ GDB log for mapper -----------------------------------------------------------

(gdb) bt
#0  __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:47
#1  0xb6b31648 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#2  0xb6b3234a in __GI_abort () at abort.c:89
#3  0xb6b2c9b2 in __assert_fail_base (fmt=0x1 <error: Cannot access memory at address 0x1>,
    assertion=0xb35a8a58 "(reinterpret_cast<size_t>(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && \"this assertion is explained here: \" \"http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAss"..., assertion@entry=0x2 <error: Cannot access memory at address 0x2>,
    file=file@entry=0xb19ff420 "\001", line=109, line@entry=3065995392,
    function=function@entry=0xb35a6790 "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:92
#4  0xb6b2ca4a in __GI___assert_fail (assertion=0x2 <error: Cannot access memory at address 0x2>, file=0xb19ff420 "\001", line=3065995392,
    function=0xb35a6790 "Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = float; int Size = 4; int MatrixOrArrayOptions = 0]") at assert.c:101
#5  0xb356014e in mapper::MapperNodelet::PclCallback(boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&) ()
   from /opt/astrobee/lib//libmapper.so
#6  0xb3581dd8 in boost::detail::function::void_function_obj_invoker1<boost::function<void (boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&)>, void, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> >::invoke(boost::detail::function::function_buffer&, boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const>) () from /opt/astrobee/lib//libmapper.so
#7  0xb35876ae in ros::SubscriptionCallbackHelperT<boost::shared_ptr<sensor_msgs::PointCloud2_<std::allocator<void> > const> const&, void>::call(ros::SubscriptionCallbackHelperCallParams&) () from /opt/astrobee/lib//libmapper.so
#8  0xb6ecbbbc in ros::SubscriptionQueue::call() () from /opt/ros/kinetic/lib/libroscpp.so
#9  0xb6e912e4 in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*) () from /opt/ros/kinetic/lib/libroscpp.so
#10 0xb6e9185a in ros::CallbackQueue::callOne(ros::WallDuration) () from /opt/ros/kinetic/lib/libroscpp.so
#11 0xb6fb4d9a in nodelet::detail::CallbackQueue::callOne() () from /opt/ros/kinetic/lib/libnodeletlib.so
#12 0xb6fb63de in nodelet::detail::CallbackQueueManager::workerThread(nodelet::detail::CallbackQueueManager::ThreadInfo*) ()
   from /opt/ros/kinetic/lib/libnodeletlib.so
#13 0xb6a6bd60 in ?? () from /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.58.0
#14 0xb6a3f5b4 in start_thread (arg=0x0) at pthread_create.c:335
#15 0xb6b9fc4c in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:89 from /lib/arm-linux-gnueabihf/libc.so.6

bcoltin, 2022-02-28:
When we upgrade to Ubuntu 20, we should also upgrade C++ version. Then we should be able to remove this extra flag and alignment will still work.

bcoltin, 2022-11-07:
Hasn't caused any problems, won't be with new Eigen versions. Closing.

closed 2022-11-07 by bcoltin