nasa/astrobee #292: Unable to launch the sim in Docker from run.sh

tammerb, 2021-09-27:
I get the following rosout ERRORs:

`[ERROR] [1632774422.100197005, 26.992000000] : (ros.Astrobee) graph_localizer_intializer.cc:47: SetBiases: Failed to create imu bias output file.`

`[ERROR] [1632774449.542471427, 54.392000000] : (ros.nodelet) Failed to load nodelet [/astrobee_astrobee_bridge] of type [dds_ros_bridge/AstrobeeAstrobeeBridge] even after refreshing the cache: Could not find library corresponding to plugin dds_ros_bridge/AstrobeeAstrobeeBridge. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`
`[ERROR] [1632774449.542554093, 54.392000000] : (ros.nodelet) The error before refreshing the cache was: Could not find library corresponding to plugin dds_ros_bridge/AstrobeeAstrobeeBridge. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`
`[FATAL] [1632774449.542753551, 54.392000000] : (ros.Astrobee./sys_monitor) Unable to load nodelet astrobee_astrobee_bridge in mlp_multibridge manager.`
`[ERROR] [1632774449.542811885, 54.392000000] : (ros.Astrobee) Fault with id 51 occurred. Fault error message: Never received heartbeat from and unable to load astrobee_astrobee_bridge
[ERROR] [1632774449.594596321, 54.440000000] : (ros.nodelet) Failed to load nodelet [/dds_ros_bridge] of type [dds_ros_bridge/DdsRosBridge] even after refreshing the cache: Could not find library corresponding to plugin dds_ros_bridge/DdsRosBridge. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`
`[ERROR] [1632774449.594674863, 54.440000000] : (ros.nodelet) The error before refreshing the cache was: Could not find library corresponding to plugin dds_ros_bridge/DdsRosBridge. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.`
`[FATAL] [1632774449.594824446, 54.440000000] : (ros.Astrobee./sys_monitor) Unable to load nodelet dds_ros_bridge in mlp_communications manager.`
`[ERROR] [1632774449.594897154, 54.440000000] : (ros.Astrobee) Fault with id 33 occurred. Fault error message: Never received heartbeat from and unable to load dds_ros_bridge`

marinagmoreira, 2021-09-27:
Which OS are you running the docker container from? Is it an Virtual Machine?

tammerb, 2021-09-27:
From my local Ubuntu 18.04.

marinagmoreira, 2021-09-29:
Sorry for the delay, it seems that the major problem is:
graph_localizer_intializer.cc:47: SetBiases: Failed to create imu bias output file

This file is created at runtime when the simulation starts, some people have problems because of file permissions in docker. A solution is to add this file before running the simulation. We're trying to get some more permanent solution for this problem soon, but for now you can append to the end of the dockerfile (scripts/docker/astrobee.Dockerfile):
RUN echo "-9.82371e-13,-1.13071e-12,-1.39643e-12 \n-1.80458e-08,-2.20222e-11,-9.01996e-13" > /src/astrobee/astrobee/resources/imu_bias.config

marinagmoreira, 2021-09-29:
TODO: @rsoussan Can you check if we can add this file to the repo, or should we add the command to the dockerfile?

tammerb, 2021-10-05:
@marinagmoreira that extra line does create the imu_bias.config file as root, but when starting the container as astrobee the same errors pop up. I fixed this by also appending ` && chmod a+rw /src/astrobee/astrobee/resources/imu_bias.config`  after your command to create the file.

bcoltin, 2021-11-01:
Marina will add imu_bias config to repo or change docker permissions.

marinagmoreira, 2021-11-04:
This should be fixed now with the latest merged PR. I tested this with my personal native ubuntu 20 computer.

closed 2021-11-04 by marinagmoreira