nasa/astrobee #310: Arm command dictionary does not match the robot

jsbarlo, 2021-11-01:
Under Arm.armPanAndTilt in the command dictionary https://github.com/nasa/astrobee/wiki/Command-Dictionary#command_Arm.gripperControl, the range of the arm tilt is listed as -30 <= val <= 90 values between -30 and -20 give an error
[image]
[image]
.

trey0, 2021-11-01:
The canonical location of the command dictionary is [1], which appears to be correct. Not sure where that wiki page came from?

[1] https://nasa.github.io/astrobee/html/command_dictionary.html#command_Arm.armPanAndTilt

bcoltin, 2021-11-01:
We will delete the wiki.

marinagmoreira, 2021-11-02:
wiki deleted with link to actual documentation

closed 2021-11-02 by marinagmoreira