nasa/astrobee #329: Merge bag tools into one

marinagmoreira, 2021-11-09:
There is currently 2 merge bag tools in astrobee/tools/graph_bag/scripts/merge_bags.py and astrobee/localization/localization_node/tools/merge_bags.cc.

The functionality of both should be merged into the c++ api one (because the python api crashes with messages from rosjava) and moved to the astrobee/tools folder for better visibility of users

bcoltin, 2022-01-20:
@rsoussan this isn't super urgent, but do you use tools/graph_bag/scripts/merge_bags.py? What does this have that astrobee/localization/localization_node/tools/merge_bags.cc lacks?

rsoussan, 2022-01-21:
It automatically finds bags with a given prefix in a directory and merges them, and adds an option to only save localization topics to keep the bags smaller.

bcoltin, 2022-02-28:
We'll move all the bag processing scripts from both locations into a new ros package in tools.

bcoltin, 2022-08-22:
We think the python script does everything and we can now delete the C++ script. However, let's wait a bit to make sure we still don't need it.

bcoltin, 2022-08-22:
@marinagmoreira will add a warning to the C++ version that it is obsolete.

marinagmoreira, 2022-10-04:
Also merged both python merge scripts into one, that can merge entire directory or selected files

bcoltin, 2022-11-08:
Will call the C++ script from python and get rid of the duplicate functionality in the python script.

closed 2023-11-06 by marinagmoreira