nasa/astrobee #331: Add bag migration rules

marinagmoreira, 2021-11-11:
When message definitions change, old rosbag files become unreadable, to solve this migration files can be defined to maintain readability. Example: flight mode was recently updated
http://wiki.ros.org/rosbag/migration

 TODO: add flight mode migration file, serving too as a template

trey0, 2021-11-12:
That is a great point.

I didn't realize we had been allowing ourselves to change message definitions in a way that could invalidate old bags. That seems like a serious problem. We now have a lot of old bags with precious ISS data that we can use for validating future changes to the software, and it would be a shame to accidentally break our ability to use them. Probably any PR that is likely to invalidate old bags should state that explicitly up front, and we should weigh carefully whether it's worth it.

Migrations are one way to help manage message schema versioning. Another (possibly complementary) approach is to use explicit schema versioning so it is obvious (like from the *.msg filename and possibly also the topic names) whether the bag contains v1 or v2 schema messages. Some messaging systems like Google Protocol Buffers have built-in support for managing versioning. Even just tweaking topic names slightly when a definition changes, whether or not the new name includes a marker like "v2", could avoid confusion later.

Those kinds of techniques may or may not be worth it in our system, but we should recognize that the existence of rosbag migrations as a tool isn't a complete solution because: (1) it may not be obvious what migration rules need to be run depending on when the bag file was captured, which could be a big headache to manage, (2) depending on the nature of the schema change, there might not be any straightforward way to migrate the old messages (like if older messages are missing a field that the newer software relies on).

This might be a good topic to discuss at an upcoming FSW meeting. It would be good to agree on guidelines for how to manage message schema changes.

bcoltin, 2022-01-20:
I'm going to close this, we've discussed mitigation strategies, we can revisit when we have a specific message change we need this for.

closed 2022-01-20 by bcoltin