nasa/astrobee #332: What's the current status of the Vive localization code

MaxPalmer-UH, 2021-11-12:
I have a use case where I'd like more accurate static pose (position and rotation) data for Vive (actually Tundra) trackers than is provided by standard tracking routines. Reading this paper (https://ti.arc.nasa.gov/publications/61557/download/) gave me hope that this was possible and that the code had been made available. 

The paper also suggests that further improvements were possible post publication.

1. Would using the Vive localisation code here give better static pose accuracy?
2. Have further improvements been made since the paper was published?
3. Is the code in a reusable form? A closed issue indicates it might be heavily tied to the experimental setup.
4. I'd like to think it would work with Tundra trackers?
5. What does the low level driver do? Does it provide access to the LED illumination states which can then be processed?
6. Does this work with Steam VR 2.0 tracking and >2 base stations? Should I expect any improvements in performance?

Thanks.

Max

bcoltin, 2021-11-15:
Hi Max,

Unfortunately the person who did most of this work left NASA, and I don't know enough to answer most of your questions. We haven't done any further work on Vive for a couple years, so likely doesn't support the latest devices, or at least has not been tested with them. I believe the code is not too tightly tied to our setup--- I believe a number of other people have been able to make use of it. Hope this helps!

DrMaxP, 2021-11-17:
Thanks for getting back to me. Would be interested in hearing from others if they have used the Vive tracker code.

bcoltin, 2021-11-17:
Sure, thank you!

closed 2021-11-17 by bcoltin