nasa/astrobee #353: *catkin2* not found

ana-GT, 2021-12-02:
Hi, I just pulled the latest version of the astrobee repository and I am having trouble building the workspace. I tried using both master and develop branches, and followed the instructions [here](https://nasa.github.io/astrobee/html/install-nonNASA.html)

1. Master branch: Running ./configure.sh --with-arguments-as-in-the-instructions , I get a message saying: --w is not a valid option"
2. Develop branch: Configures fine, but compiling fails due to *catkin2* not being found.

If you have any pointer/advice, please let me know, thanks!

marinagmoreira, 2021-12-02:
The develop branch now compiles with catkin, please re-read the install instructions. After running the configure.sh, we need to re-source ./bashrc.

The documentation updates with the develop branch, for 'master' instructions you have to read the instructions in the repo doc/general_documentation/INSTALL.md

marinagmoreira, 2021-12-02:
Forgot to mention, from the configure.sh this line is added to the bashrc (exact path depends on your setup):
if [[ ":$CMAKE_PREFIX_PATH:" != *":/home/mgouveia/ros_ws/astrobee/src/cmake:"* ]]; then CMAKE_PREFIX_PATH="${CMAKE_PREFIX_PATH:+"$CMAKE_PREFIX_PATH:"}/home/mgouveia/ros_ws/astrobee/src/cmake"; fi

this is what adds the folder where the catkin2Config.cmake is defined to the prefix path

ana-GT, 2021-12-06:
Thanks a lot Marina, sorry for the question! Re-read and re-followed the instructions and it works great as usual.

marinagmoreira, 2021-12-06:
Sounds good, maybe we can add some warnings to the documentation to make it more clear

closed 2021-12-06 by marinagmoreira