nasa/astrobee #354: catkin build: CMake Error: The source directory does not appear to contain CMakeLists.txt.

ablaghaleb, 2021-12-02:
Hi!

I am currently following the instructions [from this page](https://nasa.github.io/astrobee/html/install-nonNASA.html) to get the Astrobee repo running (master branch). However, when I run catkin build, I run into the following error: `Errors     << catkin:cmake /home/abla/astrobee/logs/catkin/build.cmake.001.log
CMake Error: The source directory "my-home/astrobee/src/cmake" does not appear to contain CMakeLists.txt.
`
I made sure to source bashrc before running catkin build. One thing of note is that my astrobee_build and astrobee_install directories are alongside the astrobee directory, not inside of it -- but I don't know if that has anything to do with the issue.

Any pointers as to what might be wrong?

Thank you!
Abla.

marinagmoreira, 2021-12-02:
Hi Abla!
Those instructions are for the develop branch, we recently switched to catkin compile. We're actually doing a release to master this week, but in the meanwhile if you use the develop branch you should be fine.
Let me know if you're still having issues after switching branches

Pedro-Roque, 2021-12-08:
@ablaghaleb it should work as of today - tested on Ubuntu 20, no issues at :)

closed 2021-12-16 by bcoltin