nasa/astrobee #356: Ubuntu 20 setup issues

Pedro-Roque, 2021-12-06:
@marinagmoreira 

I am setting up the simulator on a Ubuntu 20 machine, and the OpenCV build clashes with ROS. Namely, "libvtk6-dev" clashes with it and wants to uninstall it. 

I'll try to find a fix for it and report back here - unless you have a fix already? :)

marinagmoreira, 2021-12-06:
We build opencv now with debbuilder (in the install debians script), no need to install it from source.

We've ran into similar issues when both were trying to be installed at the same time (making the debian while having opencv installed from source), my advice is to start from a clean machine if possible and follow the instructions closely not to install opencv from source.

Pedro-Roque, 2021-12-07:
Hi @marinagmoreira, you are right, changing the installation order fixed it! 

However, when running
`./src/scripts/configure.sh -l -F -D -p $INSTALL_PATH -w $WORKSPACE_PATH` , I get the error that "w" is an illegal option. Upon inspecting the configuration script, I see that there is no option for w - is this deprecated?

Cheers,
Pedro

Pedro-Roque, 2021-12-08:
Thanks @marinagmoreira , it worked with the new release. Also, the -mtune and -march flags were nto a problem anymore, so I'm closing this issue :)

closed 2021-12-08 by Pedro-Roque