nasa/astrobee #372: pico_proxy is launched by default

oleg-alexandrov, 2021-12-22:
Running the command:

roslaunch astrobee astrobee.launch mlp:=local llp:=disabled nodes:=framestore output:=screen

launches the pico_proxy node automatically, and there seems no way to prevent it from launching.  It launches even when the pico_driver itself does not launch, when it is not of much help, as pico_proxy processes the output of pico_driver. 

While we want pico_proxy on the bot,  general users won't be able to build pico_proxy as the documentation suggests configuring and then building with " -F" which turns off building Picoflex and hence pico_proxy. This tool is also some proprietary thing and will work only with some magic private number inserted in a config file. 

I think the current logic, which is here: https://github.com/nasa/astrobee/blob/develop/astrobee/launch/robot/MLP.launch#L390 and says: 

 <include file="$(find pico_driver)/launch/pico_proxy.launch" />

should maybe modified by having a conditional somewhere, like other logic in there, so if pico_driver is not launched, pico_proxy should not be launched either.

marinagmoreira, 2021-12-22:
it's still inside the 'drivers' group in the launchfile so it will not launch unless we're using it on the robot, which uses pico_proxy

oleg-alexandrov, 2021-12-22:
So I guess one can use:

roslaunch astrobee astrobee.launch mlp:=local llp:=disabled nodes:=framestore output:=screen drivers:=false

to not get errors about missing pico_proxy, if not compiled. It would still make more sense for pico_proxy to not launch without pico_driver, but I can live with this, and will adjust the doc to exclude the drivers where such a command is needed to do things which are unrelated to depth camera.

marinagmoreira, 2021-12-29:
ohh okok I see what's the problem here! For background the pico_driver and pico_flex are under the same place on the launchfile, so they always launch together. However, the picoflex does not have the argument nodes, for it to exclude if only the framestore is desired!

marinagmoreira, 2021-12-29:
I'll check how to add these arguments to the picoflex driver in the launch file and that should totally solve your problem :)

closed 2022-01-11 by marinagmoreira