nasa/astrobee #384: Look into why the sci cam wasn't able to recover

kbrowne15, 2022-01-04:
During the Soundsee data 3 activity, the sci cam stopped working after roughly 50 images. The apk was able to detect it but unable to recover. Usually a take picture or restart continuous picture taking fixes this but during the activity, the camera returned an error saying the camera was busy. Need to look into this and try to run the scicam during the next CM.

trey0, 2022-01-04:
Thanks for filing the issue! But it should probably be filed on the https://github.com/nasa/astrobee_android repo instead, right?

bcoltin, 2022-01-04:
It is for an issue in that repository, but I'd personally prefer to keep it here because I doubt I will remember to look at the issues over there.

trey0, 2022-01-04:
> It is for an issue in that repository, but I'd personally prefer to keep it here because I doubt I will remember to look at the issues over there.

Great point!

It might be better to form the habit of looking at an overview of issues across all of the Astrobee/ISAAC repos rather than just following the link to the `astrobee` repo issues specifically.

I figured out a search query below that covers all the repos I know about, and it currently shows 24 open issues vs. just 18 for `astrobee`. Maybe you could bookmark that?

For individual developers, a more efficient heuristic might be to look at their personal issue dashboard for issues that are assigned to them. That approach automatically works across repos.

https://github.com/issues?q=is%3Aopen+is%3Aissue+archived%3Afalse+repo%3Anasa%2Fastrobee+repo%3Anasa%2Fastrobee_android+repo%3Anasa%2Fastrobee_media+repo%3Anasa%2Fastrobee_gds+repo%3Anasa%2Fisaac+repo%3Anasa%2Fisaac_msgs+repo%3Anasa%2Fisaac_user_interface+

bcoltin, 2022-01-04:
Oh nice, I didn't know you could do that!

kbrowne15, 2022-01-04:
Moved to android repo.

closed 2022-01-04 by kbrowne15