nasa/astrobee #385: graph_localizer is missing teaserpp deps in 16.0

albee, 2022-01-04:
The 16.0 release of graph_localizer is missing teaserpp dependency statements in the CMakeLists.txt that is causing the nodelet to crash in simulation. See screenshot below. (Ryan and I saw a similar issue on hardware that we managed to fix by adding a few target_link_libraries statements and catkin dependencies for teaserpp.)

Wanted to make sure this wasn't just me, if so then I don't think the sim will work for anyone with the current release. I can do a PR if needed.

[image]

albee, 2022-01-04:
I've fixed this locally by the way by adding the deps back in to the CMakeLists.txt.

rsoussan, 2022-01-06:
Which dependencies fixed the issue?  I'll test sim on my laptop, although I think I did before the main Depth Odom pr which is currently on dev.

bcoltin, 2022-01-20:
Seems like this is just Keenan and he fixed it. If that's not the case please re-open.

closed 2022-01-20 by bcoltin