nasa/astrobee #399: Mapper Struggling During CM

kbrowne15, 2022-01-25:
It has been noted that running the mapper with nice 10 will crash the mapper. With nice of 8, it seems to struggle meaning heartbeat faults for the mapper appear and disappear. We might need to try a lower nice value.

trey0, 2022-01-25:
It might be good to re-examine our strategy for allocating CPU to MLP services.

As things stand, the MLP CPU is usually over-subscribed overall, which leaves each thread with an effective CPU allocation that is less than it would use in an unloaded system. Say the sparse mapper would use 100% of a core in an unloaded system, but its effective allocation under "typical" system load is 50% of a core and it needs to have at least 20% of a core to have acceptable performance (keep localization fix, avoid crashing or triggering heartbeat faults).

When you set the nice level of a thread to NI, the run time should be reduced by about a factor of 1.25 ** NI (relative to other processes) [1]. At nice 10, it should get about 9.3x less run time. At that extreme level, it makes sense that the mapper might get starved under typical system load.

The trouble is, the correct value to use for nice is dependent on the load imposed by the other processes. Assuming the example numbers above are correct (50% typical vs. 20% min acceptable), you can calculate a target nice value of NI = log(0.5/0.2)/log(1.25) = ~4. But any new loads introduced on the MLP (such as from the ROAM guest science or future localizer changes) may reduce the typical number below 50%, requiring us to lower the target nice value or risk starving the mapper again.

An alternative way to limit CPU usage is the cgroups mechanism. It would allow us to explicitly set a limit, like 20% of a core, that doesn't depend on the other loads on the system. Might be worth investigating.

[1] https://man7.org/linux/man-pages/man7/sched.7.html
[2] https://linuxhint.com/limit_cpu_usage_process_linux/

marinagmoreira, 2022-02-08:
Thanks for looking into this Trey! Based on your feedback and lab testing the nice value was adjusted to 5.
I'll close this issue for now.

closed 2022-02-08 by marinagmoreira