nasa/astrobee #423: Simulation Drag Model not Receiving Correct Link Velocity

albee, 2022-02-22:
The drag plugin, `simulation/src/gazebo_model_plugin_drag/gazebo_model_plugin_drag.cc`, is receiving incorrect inertial frame velocity values for the body link. It seems to always get a unit vector, resulting in large computed drag values equivalent to Astrobee zooming around at 1 m/s all the time, ~0.05 N, (~1/3 of typical per-axis thruster input, for reference). Normal drag values should be about 3 orders of magnitude lower for typical Astrobee speeds, so pretty negligible. We initially noticed this when using a mass parameter estimator that was having trouble converging and always estimating high in simulation---when removing the drag plugin altogether it converges perfectly.

I've gotten this to happen locally for both my FSW 15.1 workspace and a clean `develop` workspace pulled today. I'm still waiting for someone else on our team to confirm this isn't just a weird local problem I'm having.

To pinpoint the issue, line 82 of that file above is giving bad velocity values for `GetRelativeLinearVel()`. Here are some typical values I'm seeing in sim:

[image]

I can take a look at this deeper it we confirm it's not just me seeing the issue. (Also note that the drag bounces around between negative and positive when Astrobee's actual velocity is low, which is why it's not running off from a constant drag force being applied.)

albee, 2022-02-22:
@Pedro-Roque In case he gets the chance to check it out.

Pedro-Roque, 2022-02-22:
Thanks @albee for finding this out! I'll check it on my system by the end of the week too

trey0, 2022-02-22:
Yeah, thanks @albee, nice catch!

A couple of minor followup points:
- This particular issue could be especially problematic because it accidentally provides a strong artificial stabilizing force for holding position, which could mask controller instability.
- As a non-urgent suggestion, we might want to include a term for air flow. The ISS requirement for cabin air flow is 15-40 feet per minute, and most USOS modules have overhead-to-deck as the nominal flow direction. A very simple improvement to the drag model would be to add a steady flow rate from overhead to deck at 20 feet per minute (7.6 cm/s). The real air flow is of course more complicated and variable, but even a simple term like that would provide qualitatively more realistic behavior when the robot idles propulsion--it won't hold position but instead will systematically drift, usually toward the deck.

Pedro-Roque, 2022-02-22:
> Yeah, thanks @albee, nice catch!
>
> A couple of minor followup points:
>
> * This particular issue could be especially problematic because it accidentally provides a strong artificial stabilizing force [...]

@trey0 , on the ReSWARM session #2 I noticed a significant difference between simulation and ISS behavior, that now I believe could come from this problem. I will check on my code if the bug is present (since I use the same stack as @albee , most likely it will). I also have / can have bags from Simulation and ISS Deployment for unit tests in case they are helpful for anything?

trey0, 2022-02-22:
Hi @Pedro-Roque,

You can ask Jose Benavides about data availability in general. In very recent news, the Astrobee Facility has just publicly released Astrobee ISS telemetry bags for the first time [1]. So far it covers two ISS activities, hopefully with more on the way. It's a work in progress.

[1] https://www.nasa.gov/content/guest-science-resources - follow the "Data" link at the bottom

marinagmoreira, 2022-02-23:
I tested in noetic and while moving at max speed:
drag_.Length(): 0.168776
drag_.Normalize(): 0.058454 0.996875 0.053141

marinagmoreira, 2022-02-23:
I can confirm this problem on kinetic:
drag_.GetLength(): 1
drag_.Normalize(): 0.891642 -0.327231 -0.31288

albee, 2022-02-23:
@marinagmoreira Thanks, @Pedro-Roque and I are going to look for a fix next week. @trey0 we should be able to add in that constant airflow term too, we'd just have to differentiate between the ISS and granite sim worlds.

trey0, 2022-02-23:
>  we should be able to add in that constant airflow term too, we'd just have to differentiate between the ISS and granite sim worlds.

Gotcha. You may have more insight here than I do, but maybe we need not bother to differentiate? I assume/hope that a slight downdraft in the granite sim would not cause the robot to push through the granite table!

Pedro-Roque, 2022-03-24:
Hi @marinagmoreira,

I've tested the drag model on 20.04, and the problem persists - the drag is much higher than what we believe it should be. Here is a print of the debug points:
[image]

and here is a print of the output, with the Astrobee being static in the station, after undocking:
[image]

So it seems that some of the constants are, then, not great? I'll investigate a bit more...

marinagmoreira, 2022-03-24:
So you think the velocity is fine, but the coefficient/area/density might be off? Or is the velocity also wrong?

Pedro-Roque, 2022-03-24:
@marinagmoreira I believe the issue is the in-situ modification of drag_.Normalize(), which then drag_.Length() picks-up as "1" and not as the real velocity.

Adding an extra variable for the velocity fixed the issue, as far as we can see.

Pedro-Roque, 2022-03-24:
Yep, thanks @albee !! :smile:

albee, 2022-03-24:
Fixed by #462.

Thanks @marinagmoreira  @Pedro-Roque @trey0 !

closed 2022-03-24 by albee