nasa/astrobee #429: Bags for public release should pass "rosbag check"

trey0, 2022-02-28:
When we release a bag, we should do our best to minimize pain points for external users trying to analyze the data. Here are some known issues to address:

- [ ] We should fix missing message definitions per [1]. Marina found a fixer script with usage described at [2]. However, it currently doesn't work right on bags that contain obsolete message definitions.
- [ ] We should consider proactively migrating the messages in the bag to the latest message definitions, so they can be analyzed using the latest FSW build. (Or figure out a better approach.)

As a reminder, when these problems exist in the bag you can easily get the rosbag tools crashing with a confusing bogus error message, even if you are only trying to process a different message that doesn't have the problem.

[1] https://github.com/nasa/astrobee/issues/402
[2] https://github.com/nasa/astrobee/blob/develop/doc/general_documentation/public_data.md

marinagmoreira, 2022-03-10:
I'm not sure what you mean with "doesn't work right on bags that contain obsolete message definitions", it works for me using the command:

````
python fix_bag_msg_def.py --use-local-defs /home/mgouveia/data/bags/original/20210326_1653_phase1Loc_survey_bay_6_attempt_2.bag out.bag
````

it does generate warnings:

````
Writing out fixed bag ..
 out.bag                                                                                                                                                                                                 0%   0.0 KB /  32.0 GB --:-- ETAWARNING: For type [ff_msgs/FlightMode] stored md5sum [21af9f940eeb9407d4d58829e8cfc200] does not match message definition [0bb389101a5f30087bd644e6596d8e8e].
  Try: "rosrun rosbag fix_msg_defs.py old_bag new_bag."
WARNING: md5sum of loaded type [ff_msgs/FlightMode] does not match that specified
 out.bag                                                                                                                                                                                                 3%   1.1 GB /  32.0 GB 03:25 ETAWARNING: For type [isaac_msgs/InspectionActionGoal] stored md5sum [d36308676d466059465a3cf2963a88fe] does not match message definition [b28a9e00b25932317b0c4fe9c202f1c2].
  Try: "rosrun rosbag fix_msg_defs.py old_bag new_bag."
WARNING: md5sum of loaded type [isaac_msgs/InspectionActionGoal] does not match that specified
 out.bag                                                                                                                                                                                                 3%   1.2 GB /  32.0 GB 03:26 ETAWARNING: For type [isaac_msgs/InspectionActionResult] stored md5sum [89865434ba50475f32f54cd1316ada77] does not match message definition [76d3975d48ce46983c64db488a0f3363].
  Try: "rosrun rosbag fix_msg_defs.py old_bag new_bag."
WARNING: md5sum of loaded type [isaac_msgs/InspectionActionResult] does not match that specified
 out.bag                                                                                                                                                                                                73%  23.6 GB /  32.0 GB 01:01 ETAT
 ````
 but doesn't fail

trey0, 2022-03-10:
> I'm not sure what you mean with "doesn't work right on bags that contain obsolete message definitions"

When doing this kind of message definition rewriting, it's easy to accidentally leave the bag in an inconsistent state where the message definition doesn't match the binary message data, so you get binary corruption garbage when you try to deserialize (or an error message if you're lucky).

Based on the warnings you mentioned, I was worried that would be the case here. Admittedly I didn't check how bad it was, just decided I didn't trust the output.

trey0, 2022-03-11:
Closed by #447 

Maintaining telemetry backward compatibility will be an ongoing process. See strategy at:
https://github.com/nasa/astrobee/blob/develop/doc/general_documentation/maintaining_telemetry.md

commit bbe0dc03 (2022-03-11): Enable backward compatibility with archived telemetry bags (#447)

Fixes #429
closed 2022-03-11 by trey0