nasa/astrobee #438: Boolean config parameters not working on 20.04

marinagmoreira, 2022-03-04:
config:
 {
    id = "disable_mapper",
    reconfigurable = false,
    type = "boolean",
    default = true,
    unit = "boolean",
    description = "Should we disable the mapper?"
  },{
    id = "enable_obstacles",
    reconfigurable = false,
    type = "boolean",
    default = false,
    unit = "boolean",
    description = "Should we enable obstacles?"
  }
 
code:

    ROS_ERROR_STREAM("disable_mapper!" << cfg_.Get<bool>("disable_mapper"));
    ROS_ERROR_STREAM("enable_obstacles!" << cfg_.Get<bool>("enable_obstacles"));
 
result
[ERROR] [1646354194.403604700, 0.576000000] : (ros.mapper) disable_mapper!0
[ERROR] [1646354194.421923449, 0.592000000] : (ros.mapper) enable_obstacles!1

marinagmoreira, 2022-03-05:
It seems that in 20.04 the rosparam are persistent. Even after killing the simulation, if you do `rosparam list`, you will see all the parameters.
This is problemetic because in the config server there are the commands:
```` 
     nh_.getParam(id, bools_[id]);
     nh_.setParam(id, bools_[id]);
````

bcoltin, 2022-03-07:
Can you check if roscore is still running? I believe these are stored in roscore.

Although either way it is probably better to move these into the config.

marinagmoreira, 2022-03-08:
> Can you check if roscore is still running? I believe these are stored in roscore.
> 
> Although either way it is probably better to move these into the config.

Yeah it might have been too that a roscore was running somewhere else! Given that it is not expected behavior that the config is overwritten by the rosparam still a good idea to take it out :)

closed 2022-03-08 by marinagmoreira