nasa/astrobee #445: Add image metadata to recorded bagfiles

rsoussan, 2022-03-08:
It would be helpful to record some of the image configurations, such as gain and exposure, during activities for future analysis.  This could be generalized for other important config files as well if needed.

bcoltin, 2022-03-08:
We had also talked before about publishing a "CameraInfo" message with the calibration data.

trey0, 2022-03-08:
Both of these make sense, but I think we are talking about different things:
1. Calibration info which is hopefully static fits perfectly into the standard ROS `sensor_msgs/CameraInfo` message. (Typically publish once, latching.)
2. We might also want to publish information that has the potential to be dynamic on a per-image basis. Examples would be focus, exposure, and white balance parameters that are often auto-adjusted dynamically by the camera or the driver. These are the kinds of things that camera hardware normally saves in the photo metadata (relevant metadata standards include EXIF, IPTC, and XMP). I'm not sure if ROS has established a standard message for that. Currently, we try to turn off most of the "auto" features for NavCam/DockCam, but I doubt we got everything. For example, if auto-white-balance is enabled, but we could record the color temperature changes on the fly, we could correct for the changes across multiple images when building maps and avoid color casting artifacts. And if we want to have the option of turning on auto-exposure in the future, Ryan already noted that if we publish the exposure parameters, that could help the localization maintain feature tracking through exposure changes.

bcoltin, 2022-03-08:
Yeah, I agree, just figured we might want to do calibration if we're going to do these other parameters.

marinagmoreira, 2022-03-14:
Are all the parameters defined in rosparam? (I know that parameters that use config_server are) (https://github.com/nasa/astrobee/blob/master/shared/ff_util/src/config_server/config_server.cc#L177)

We could easily add a 'rosparam dump' to the logs (maybe by adding it to the data_bagger when starting the recording)

marinagmoreira, 2022-03-15:
Could we change the config reader to publish to rosparam and record it? Might be a generalized solution that's future proof.

bcoltin, 2022-03-15:
I think the issue would be with Trey's #2 point above, parameters that can change dynamically.

bcoltin, 2022-05-16:
We have gain and exposure now but could use a more general metadata message

bcoltin, 2022-08-22:
We may want to add a more general message in the future (or add more information to the current message).

closed 2022-08-22 by bcoltin