nasa/astrobee #451: Message definition field deleted without analyzing impact

trey0, 2022-03-15:
In #446 the `ff_msgs/FlightMode` message definition changed to delete the `collision_radius` field.

Whenever a PR changes a message definition in a way that could break backward compatibility, that fact should probably be highlighted in the PR description so that we can decide if it's ok according to the telemetry maintenance guidelines [1]. A migration rule should also be created, if needed.

Deleting a field specifically can be problematic, because migrating instances of the legacy message to the new message definition will necessarily lose data. (But it may be ok if the field was never useful for anything in the first place.)

@kbrowne15 Maybe we should revert this message definition change? Or maybe you know the field was never useful for anything in the archived telemetry bags?

[1] https://github.com/nasa/astrobee/blob/develop/doc/general_documentation/maintaining_telemetry.md

kbrowne15, 2022-03-22:
The collision radius was never useful for anything in the archived telemetry bags. The value was initially read out of the world config file by the choreographer but was never used by the choreographer. The choreographer is the node that publishes the flight mode message. The mapper is the only node in the system to use the collision radius and the only node that configures the mapper with this value is the executive. So if one does need to look at the collision radius from archived telemetry bags, they should look at the collision distance in the agent state stamped message published by the executive.

trey0, 2022-03-22:
Thanks for the context, @kbrowne15!

closed 2022-03-22 by trey0