nasa/astrobee #461: Can't disable keepin / keepout checks

mkhansenbot, 2022-03-24:
I'm trying to teleop the robot outside the ISS and it fails due to the keepin/keepout zones. I looked through the code and believe I should be able to set the `/cfg/choreographer/enable_validation` parameter to disable those checks but that doesn't work. It seems to keep re-setting the enable back to true:
```
/# rosrun executive teleop_tool -move -relative -pos "2 0 0"

Started simpleMove6DOF command. It may take some time to complete.

simpleMove6DOF command failed! Move goal failed with response: Keep in zone violation
/# rosparam get /cfg/choreographer/enable_validation 
true
/# rosparam set /cfg/choreographer/enable_validation false                                           
/# rosparam get /cfg/choreographer/enable_validation 
false
/# rosrun executive teleop_tool -move -relative -pos "2 0 0"

Started simpleMove6DOF command. It may take some time to complete.

simpleMove6DOF command failed! Move goal failed with response: Keep in zone violation
/# rosparam get /cfg/choreographer/enable_validation 
true
/# 
```
I believe this should be dynamically configurable based on the choreographer readme file:
https://github.com/nasa/astrobee/blob/master/mobility/choreographer/readme.md#configurable-parameters

Is this a bug or am I misunderstanding this? Is there a way to turn off keepout/keepin zones?

cc:@limenutt who I'm helping with this

marinagmoreira, 2022-03-24:
So what's happening here is that the executive configures the choreographer every time you send a mobility command using the value it has initialized or configured through GDS (used in ops).
For now you can use the mobility tool with: (no relative move though, we can add it if needed)
`rosrun mobility teleop -move -pos "2 0 0" -novalidate`
@kbrowne15 is there a way of configuring the executive without GDS?

mkhansenbot, 2022-03-24:
That works great! Thank you for helping.

Just a note that I had also tried changing the config file for the choreographer (astrobee/astrobee/config/mobility/choreographer.config) to set the `enable_validation` default to 'false' before running the simulation and that didn't work. I think you may want to test that and see if it's working as intended. 

[image]

marinagmoreira, 2022-03-24:
I've tested the config file, it does affect how the parameter is set at startup, but it gets reconfigured every time a motion command is sent.
Note that the config file itself is not reconfigurable after startup (it doesn't pick up the edits), but it's reconfigured through a client (https://github.com/nasa/astrobee/blob/master/mobility/mobility/tools/teleop.cc#L409).

kbrowne15, 2022-03-24:
There is a script that you can use to configure the executive. To turn keepin/keepouts off, run:
`./scripts/teleop/set_check_zones.sh false`
To turn them on, run:
`./scripts/teleop/set_check_zones.sh true`

mkhansenbot, 2022-03-24:
> I've tested the config file, it does affect how the parameter is set at startup, but it gets reconfigured every time a motion command is sent. Note that the config file itself is not reconfigurable [...]

I was setting the config prior to running the simulation, so it seems to reset to a default of true every time the executive is run, regardless of the initial setting.

I'll try the `set_check_zones.sh` method mentioned above by @kbrowne15. Is that documented somewhere?

mkhansenbot, 2022-03-24:
I tried the set_check_zones method and it worked the first time I run the `executive teleop_tool` command but the second time I get a different error:
```
/# rosrun executive teleop_tool -move -relative -pos "2 0 0"

Started simpleMove6DOF command. It may take some time to complete.
pos: x: 12.81 y: -9.81 z: 4.30 att: x: -0.00 y: 0.00 z: 0.00 w: 1.00
simpleMove6DOF command failed! Move goal failed with response: Position tolerance violated
```
I'm just posting this for completeness. I don't know whether that's expected or not, but it doesn't meet my needs. However the `mobility teleop` method already worked so I don't need this. I just thought this might be useful information. I don't know whether that is expected or not. 

If all of this is expected behavior, then you can close this issue. Thanks for the help.

kbrowne15, 2022-03-24:
The set_check_zones.sh method is not documented. It was meant for internal testing. I'll work on adding it to the teleop tool so it is easier for external users to set.

closed 2022-05-16 by bcoltin