nasa/astrobee #465: ci_push probably not testing what it should

trey0, 2022-03-25:
Both [1] and [2] have steps "Build code" and "Test code".

In [1] there is a `REMOTE` arg only for the "Build code" step but in [2] there is a `REMOTE` arg for both steps. I think [1] has the desired behavior and [2] has a subtle bug.

Basically, I think it is intending to test the current code, but instead it is testing whatever version was previously pushed to ghcr.io. But I could be wrong about the intent.

[1] https://github.com/nasa/astrobee/blob/master/.github/workflows/ci_pr.yml
[2] https://github.com/trey0/astrobee/blob/workflow_dispatch/.github/workflows/ci_push.yml

trey0, 2022-03-25:
I see what you mean. It seems like it's doing the right thing, but in a slightly obfuscated way. By comparison, I find the pattern you used in `ci_release.yml` easier to follow.

closed 2022-03-25 by trey0

marinagmoreira, 2022-03-25:
In CI release I am forced to retag everything because I need to add the version number