nasa/astrobee #532: ResourceNotFound: astrobee_iss when launching

xbbao-metis, 2022-08-17:
Howdy y'all,

I have a clean install of Ubuntu 16.04 running ROS Kinetic and astrobee. Tried just launching sim.launch and was met with the following error:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 308, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 750, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 722, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 656, in _recurse_load
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 686, in _recurse_load
    val = self._include_tag(tag, context, ros_config, default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 627, in _include_tag
    default_machine, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 666, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 257, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile', 'command'))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 207, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: astrobee_iss
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/bao/astrobee/src/astrobee
ROS path [2]=/home/bao/astrobee/src/tools/bag_processing
ROS path [3]=/home/bao/astrobee/src/communications/dds_msgs
ROS path [4]=/home/bao/astrobee/src/description/description
ROS path [5]=/home/bao/astrobee/src/shared/ff_common
...
```

Where is astrobee_iss supposed to be found? I saw that there were a couple launch files that were referencing it as well as iss_world.world - thanks!

marinagmoreira, 2022-08-17:
It seems that you might have missed the step that clones the astrobee_media repository:
git submodule update --init --depth 1 description/media
It's in https://nasa.github.io/astrobee/html/install-nonNASA.html under "Checkout the project source code"

xbbao-metis, 2022-08-18:
It looks like the checkout step is failing for some reason?
```
...
Receiving objects: 100% (152/152), 62.14 MiB | 13.83 MiB/s, done.
Resolving deltas: 100% (31/31), done.
Checking connectivity... done.
fatal: reference is not a tree: e3561545c61b6be5406839b9d959becbfe7b4100
Unable to checkout 'e3561545c61b6be5406839b9d959becbfe7b4100' in submodule path 'description/media'
```

marinagmoreira, 2022-08-18:
Given that all our unit tests/workflows pass, the checkout is not broken, but you might have given contradictory commands to put it in a bad state. There are multiple ways of fixing it https://stackoverflow.com/questions/2155887/git-submodule-head-reference-is-not-a-tree-error.

closed 2022-08-22 by bcoltin