nasa/astrobee #544: Organize the calibration code

oleg-alexandrov, 2022-10-07:
The main calibration page seems to be: https://nasa.github.io/astrobee/html/calibration.html

Yet, that is not the documentation which is in fact used for calibration. That one is at scripts/calibrate/readme.md (https://github.com/nasa/astrobee/tree/master/scripts/calibrate). I will soon make some minor style updates to this latter page and make a pull request.  

I hope these two pages can be reviewed, integrated, and the generated html can be cleaned up.

marinagmoreira, 2022-10-11:
That might also be a good place to keep a log of what is calibrated/when it was calibrated :)

oleg-alexandrov, 2022-10-11:
I read more this calibration.html doc. The formatting needs to be improved. There is also no example of target.yaml or of camera_target_based_intrinsics_calibrator.config. It is not clear how to select the right calibration target type. 

The pictures also don't show up in the html.

bcoltin, 2022-11-14:
Also add comment about how the calibration was done

closed 2023-01-30 by bcoltin