nasa/astrobee #546: qt.qpa.xcb: could not connect to display

qixifriend, 2022-10-09:
I download docker image (v0.16.6-ubuntu20.04),and run the command:
docker run -it --rm --name astrobee --volume=/tmp/.X11-unix:/tmp/.X11-unix:rw --volume=/tmp/.docker.xauth:/tmp/.docker.xauth:rw '--env="XAUTHORITY=/tmp/.docker.xauth"' '--env="DISPLAY"' --gpus all ghcr.io/nasa/astrobee:v0.16.6-ubuntu20.04 /astrobee_init.sh roslaunch astrobee sim.launch dds:=false robot:=sim_pub rviz:=true
but  display the  errs:

process[llp_imu_aug-8]: started with pid [89]
process[llp_monitors-9]: started with pid [91]
process[llp_i2c-10]: started with pid [92]
process[llp_serial-11]: started with pid [93]
process[llp_pmc-12]: started with pid [113]
process[llp_imu-13]: started with pid [119]
process[llp_lights-14]: started with pid [150]
process[imu_aug-15]: started with pid [154]
process[ctl-16]: started with pid [173]
process[fam-17]: started with pid [180]
process[mlp_localization-18]: started with pid [204]
process[mlp_graph_localization-19]: started with pid [233]
process[mlp_vision-20]: started with pid [256]
process[mlp_depth_cam-21]: started with pid [272]
process[mlp_mapper-22]: started with pid [273]
process[mlp_management-23]: started with pid [275]
process[mlp_recording-24]: started with pid [301]
process[mlp_monitors-25]: started with pid [306]
[rviz_node-3] process has died [pid 70, exit code -6, cmd /opt/ros/noetic/lib/rviz/rviz -d /src/astrobee/src/astrobee/resources/rviz/iss.rviz __name:=rviz_node __log:=/root/.ros/log/6cd58ab4-476b-11ed-a4e7-0242ac110003/rviz_node-3.log].
log file: /root/.ros/log/6cd58ab4-476b-11ed-a4e7-0242ac110003/rviz_node-3*.log
[rviz_node-3] restarting process
process[mlp_communications-26]: started with pid [307]
process[rviz_node-3]: started with pid [342]
process[mlp_multibridge-27]: started with pid [348]
process[mlp_serial-28]: started with pid [394]
_**qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.**_
why?How to fix that?Thank you and best regards.

marinagmoreira, 2022-10-18:
I can reproduce this issue, will work on it.

marinagmoreira, 2022-10-26:
The linked PR should fix this issue (tested in ubuntu 20.04) once merged

closed 2022-11-04 by marinagmoreira