nasa/astrobee #559: Don't build unused library? libastrobee_astrobee_bridge

trey0, 2022-10-31:
In `dds_ros_bridge`, the [CMakeLists.txt](https://github.com/nasa/astrobee/blob/master/communications/dds_ros_bridge/CMakeLists.txt) builds two libraries, but only one of them is actually used in [nodelet_plugins.xml](https://github.com/nasa/astrobee/blob/master/communications/dds_ros_bridge/nodelet_plugins.xml).

The other one, `libastrobee_astrobee_bridge`, appears to be unused and therefore should probably not be built to avoid confusion. (As currently specified, I think it would fail due to missing dependencies if anybody tried to use it.)

marinagmoreira, 2022-10-31:
I've raised this problem to Ruben and Katie in the past to ask if I could delete it, but I think we forgot about it

rgarciaruiz, 2022-11-21:
Resolved, PR #579

closed 2022-11-21 by rgarciaruiz