nasa/astrobee #567: NavCam, DockCam, and SciCam no images

qixifriend, 2022-11-04:
when I run simulation，
in the terminal display: 
Node::Advertise(): Error advertising topic [default/bsharp/body/nav_cam/image]. Did you forget to start the discovery service?
Node::Advertise(): Error advertising topic [default/bsharp/body/dock_cam/image]. Did you forget to start the discovery service?
in the rviz,I can see the ISS and the robot,but I can't see the images of cameras(NavCam, DockCam, and SciCam).
How to enable cameras in simulation.

qixifriend, 2022-11-04:
When running RVIZ, you may notice the big "No Image" text in the two DEBUG windows that are meant to show camera information. This is because, by default, we don't publish simulated camera images for the navigation and dock cameras due to the fact that simulating these images is very resource intensive. If you do want to generate these images, please open astrobee/config/simulation/simulation.config and change one or more of nav_cam_rate and dock_cam_rate to something other than 0. We recommend setting these values to 1 meaning these images will be published once per second. After this, you should see those images when you start the simulator. Please note this only works when the namespace is blank. In other words, please don't use the ns flag when launching the simulator if you want to see the images. To show sci camera images, modify analogously the sci_cam_rate field, and enable this camera from the Panels menu in RVIZ.

closed 2022-11-04 by qixifriend

marinagmoreira, 2022-11-04:
Hi! I'd like to add here, that you can edit that `simulation.config` file and then still start the fsw with:
`./scripts/docker/run.sh --remote --mount`
The `--mount` option will mount your src into the docker container, and since config files are read at runtime, it will acknowledge the change without need for a recompile/new image/manual sim start.