nasa/astrobee #662: Upgrade to OpenCV 4

bcoltin, 2023-01-31:
@ana-GT is looking into this with help from @marinagmoreira . We will try to merge this into both develop and ros2 branch assuming the changes aren't too drastic, supporting both versions at once. We will also need to upgrade the alvar debian.

ana-GT, 2023-02-06:
Yes! So, I was taking a look at some existing upgrades in the community and ran across this PR : https://github.com/ros-perception/ar_track_alvar/pull/89 . It is not approved yet, but from the conversation around it, it sounds it could be a good reference. I am doing some local updates in the astrobee localization code in my machine (upgrades to use OpenCV.4.x) to see if they work well together with this PR.  :crossed_fingers:

ana-GT, 2023-02-11:
Update: Was able to compile the astrobee workspace using OpenCV 4.2 (default version in noetic) and the ar_track_alvar package I mentioned in my previous message :) . I tried the test in marker_tracking (test_marker_detector.test) and it was successful. I did not test with other harder images, though.

The OpenCV4-compatible branch is in my astrobee fork and instructions on how to install things are here: https://github.com/ana-GT/astrobee/wiki/Test-OpenCV4-branch .  Other than installing OpenCV4.2 ahead and using the ar_track_alvar package from source instead of the old fork, the rest of changes are reasonably straightforward.

bcoltin, 2023-02-13:
Wonderful!!! Thank you Ana, we will check it out!

closed 2024-06-13 by marinagmoreira