nasa/astrobee #682: Make coverage_analyzer.py pipeline repeatable

trey0, 2023-02-15:
As noted in #681, the output of `coverage_analyzer.py` is not repeatable from run to run. This is not the fault of the script itself, but rather, the upstream nodes (bag playback plus localization nodes) produce variable output when the bag playback is run at 1:1 real-time speed because the localization nodes saturate and (by design) drop images that they can't process in real-time.

To give a sense of the magnitude of the problem, I found that when running on the `bartlett` server, the number of lines in the output CSV file of features increased from 63938 to 322817 (5x increase) when the `rosbag play --rate` was reduced from the default 1 to 0.1. Similar behavior could be replicated on `astrobeast`. Furthermore, even 0.1x was not slow enough to capture 100% of the data... it seems like the worst-case processing time for localization inputs has very long tails, so you get a few dropouts even if the playback rate is much slower than the average processing time.

This seems like a serious issue because the saturation level of the pipeline depends both on the computational power of the host and how loaded it is from moment to moment, so those uncontrolled factors could introduce a lot of unwelcome noise into our assessment of map quality.

@rsoussan: Do you have experience with a more robust way to operate the localization pipeline on archived telemetry that wouldn't suffer from this problem?

@amoravargas I want to make sure you're aware of this.

amoravargas, 2023-02-16:
Thanks for the feedback and keeping me in the loop.

rsoussan, 2023-02-16:
Interesting, the localizer has some buffers for holding data, and a set drop rate for images, but I would have thought with the current tuning which is for the on-robot hardware that optimization times wouldn't be a bottleneck and it would be able to handle real time just as well as slower-than-real time during an offline computation on a fast enough server.  The biggest difference would maybe be the map matching, as that can take up to 1 second on on-board hardware. Is sparse mapping node being run during this test? Sometimes when we run offline tests we don't re-run the sparse mapper and just use recorded matches, but I'm guessing for this scenario you'd like to test with different maps so that probably isn't an option. Can you link the two csv files corresponding to the real time and 0.1 time playback? 
An alternative is to use the graph_bag tool in localization_analysis, which doesn't rely on a playback speed and simply reads messages sequentially from a bagfile, with simulated tunable time delays added to mimic on-board optimization delays.

trey0, 2023-02-16:
> The biggest difference would maybe be the map matching, as that can take up to 1 second on on-board hardware. Is sparse mapping node being run during this test?

This pipeline is running in the context of sparse map coverage analysis, where the whole point is counting sparse map features matched in various parts of the module. It uses a stripped down `roslaunch` command that runs only the `framestore` and `localization_nodelet` (see the command in [1] step 2) and the analysis script listens only to the `/loc/ml/features` topic.

[1] https://github.com/nasa/astrobee/blob/master/scripts/postprocessing/coverage_analysis/readme.txt

> An alternative is to use the graph_bag tool in localization_analysis, which doesn't rely on a playback speed and simply reads messages sequentially from a bagfile, with simulated tunable time delays added to mimic on-board optimization delays.

This is more what I was thinking of. For this analysis, we want to ensure all images are fully processed by sparse map localization so we can count the features matched in a repeatable way independent of the computational power available during the analysis. The analysis doesn't pay attention to processing time and doesn't need to simulate onboard delays. Does this sound like something that `graph_bag` can do? If so, how should it be invoked, and are there any gotchas to watch out for? Thanks!

trey0, 2023-02-16:
Thanks @rsoussan!

With your feedback, I think we can table this issue for now pending how much priority the Facility team wants to put on it. (I don't think it's critical for upcoming ISAAC activities, for example.)

If we want to fix it properly, it looks like the most natural path forward would include these tasks:
- Modify the `graph_bag` tool to save the `/loc/ml/features` topic in addition to the other topics it already saves.
- Modify step 2 of the coverage analysis instructions to set up the localization pipeline in a different way. Rather than using `roslaunch` to fire up a ROS master plus flight software nodes and then running `rosbag play`, instead use the `graph_bag` tool, which doesn't need any other processes running... it just reads from an input bag and writes to an output bag. Using `graph_bag` simplifies other aspects of configuration management as well: you can specify the map file you want it to use on the command line without installing your map to the default location where FSW loads it from.
- Modify `activity_db_generator.py` to optionally take a bag as an input argument and read its feature messages from that bag instead of subscribing to them. That small change enables the script to directly read the `graph_bag` output bag and ensure it receives every message, again without firing up a ROS master and running `rosbag play`.

amoravargas, 2023-02-23:
Hi guys,

I modified locally `activity_db_generator.py` to read directly from an input bag the messages from the topic `/loc/ml/features` instead of the old method. I tested it with a few bags and it is working.

trey0, 2023-02-23:
> I modified locally `activity_db_generator.py` to read directly from an input bag the messages from the topic `/loc/ml/features` instead of the old method. I tested it with a few bags and it is working.

Great!

But just to verify we're all on the same page, I think if you use the `/loc/ml/features` topic from the *original robot bag*, it has these limitations:
- If the sparse mapping running onboard was running at ~1 Hz and dropping most of the frames, of course you won't see features from those frames (this is like the previous issue but more pronounced because the onboard processor is slower). On the other hand, many of the features you do see might come from NavCam frames that were not recorded in the bag, which is slightly confusing (e.g., image sampler recording at ~5 Hz vs. 15 Hz frame rate).
- Of course, used this way, the coverage analysis can only evaluate the map that was in use during the activity.

But it sounds easier to use than the old technique as long as we understand what it's good for, and it also sets us up to use the `graph_bag` output bag when we're ready for that.

amoravargas, 2023-02-23:
@trey0 I modified it to prepare for the change you suggested: modifying `graph_bag` and using the resulting bag as the input for `activity_db_generator.py`. I already mentioned it to @rsoussan. Once that change is done, I will test it and modify step 2 from the [Readme](https://github.com/nasa/astrobee/blob/master/scripts/postprocessing/coverage_analysis/readme.txt).

rgarciaruiz, 2023-03-14:
I believe graph_bag may be a bit too much for this specific task. Would it be possible to add a strip-down version of the tool that would only read images from a bag, run sparse mapping localize only, and then output /loc/ml/features to a different bag/csv file?

Similar to what ekf_bag/sparse_map_bag used to do, plus bag recording: https://github.com/nasa/astrobee/blob/v0.16.2/tools/ekf_bag/tools/sparse_map_bag.cc

rsoussan, 2023-03-14:
Yes, should be pretty straightforward and quick to add that - I'll try to get to it this week

trey0, 2023-03-14:
> I believe graph_bag may be a bit too much for this specific task.

I agree `graph_bag` is overkill for this use case, but unless there is some compelling need to optimize this, we should let @rsoussan focus on the onboard localization improvements in work.

marinagmoreira, 2023-07-05:
@amoravargas confirmed the merge should solve the problem

closed 2023-07-05 by marinagmoreira