nasa/astrobee #723: ROS2: OpenCV Version Incompatible

Pedro-Roque, 2023-04-17:
Hi everyone,

## Problem:

I was compiling Astrobee with colcon from scratch, natively, and I got issues regarding the `camera` package in localization, particularly:
```
/usr/bin/ld: CMakeFiles/undistort_image.dir/tools/undistort_image.cc.o: in function `main':
undistort_image.cc:(.text.startup+0x916): undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
/usr/bin/ld: undistort_image.cc:(.text.startup+0xfaf): undefined reference to `cv::imread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int)'
/usr/bin/ld: undistort_image.cc:(.text.startup+0x161c): undefined reference to `cv::imwrite(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&)'
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::UndistortCentered(Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, Eigen::Matrix<double, 2, 1, 0, 2, 1>*) const':
camera_params.cc:(.text+0x182b): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1985): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1ade): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x21eb): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::CameraParameters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
camera_params.cc:(.text+0x24f6): undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/undistort_image.dir/build.make:204: undistort_image] Error 1
make[1]: *** [CMakeFiles/Makefile2:187: CMakeFiles/undistort_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::UndistortCentered(Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, Eigen::Matrix<double, 2, 1, 0, 2, 1>*) const':
camera_params.cc:(.text+0x182b): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1985): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1ade): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x21eb): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::CameraParameters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
camera_params.cc:(.text+0x24f6): undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_camera_model.dir/build.make:203: test_camera_model] Error 1
make[1]: *** [CMakeFiles/Makefile2:215: CMakeFiles/test_camera_model.dir/all] Error 2
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::UndistortCentered(Eigen::Matrix<double, 2, 1, 0, 2, 1> const&, Eigen::Matrix<double, 2, 1, 0, 2, 1>*) const':
camera_params.cc:(.text+0x182b): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1985): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x1ade): undefined reference to `cv::Mat::updateContinuityFlag()'
/usr/bin/ld: camera_params.cc:(.text+0x21eb): undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
/usr/bin/ld: libcamera.a(camera_params.cc.o): in function `camera::CameraParameters::CameraParameters(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
camera_params.cc:(.text+0x24f6): undefined reference to `cv::FileStorage::FileStorage(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/test_camera_params.dir/build.make:203: test_camera_params] Error 1
make[1]: *** [CMakeFiles/Makefile2:243: CMakeFiles/test_camera_params.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
---
Failed   <<< camera [0.69s, exited with code 2]
```
I believe most of these errors come from a wrong OpenCV version. For instance, updateContinuityFlag() was not introduced in 3.3.1 (https://docs.opencv.org/3.3.1/d3/d63/classcv_1_1Mat.html#a3520c52cecff18ab7b243b2b45df7bc0), but was introduced in OpenCV 3.4.2 (https://docs.opencv.org/3.4.2/d3/d63/classcv_1_1Mat.html#a3520c52cecff18ab7b243b2b45df7bc0).

However the Debian build still requires 3.3.1:
https://github.com/nasa/astrobee/blob/955d6bf383cf06af90a1406cef12ab6482e7a2b1/scripts/setup/debians/build_opencv.sh#L28-L29

## Fix: 

Working on it, will post a solution here.

ana-GT, 2023-04-18:
@Pedro-Roque  It should be straightforward. I did a quick test by creating a branch from *ros2* and merged my *opencv4* branch into it (made for the **develop** branch of Astrobee),without any issues.

marinagmoreira, 2023-04-18:
I can confirm that we have a mix of OpenCV 3.3.1 (installed with astrobee) and OpenCV4.2 (I think it gets installed with the gazebo packages?). So this: https://github.com/nasa/astrobee/issues/662 will possible solve your issue eventually, or installing the same packages as we do in the docker.

Pedro-Roque, 2023-04-18:
@ana-GT I will try that on Thursday and report back here :) Thanks for checking it out!

Pedro-Roque, 2023-04-19:
@ana-GT I found some issues with your branch has libalvar still depends on 3.3.1 and there are a few patches that would be for OpenCV 3.3.1, but as I see it, there is no longer OpenCV3.3.1 in the source, right?

marinagmoreira, 2023-05-31:
> @ana-GT I found some issues with your branch has libalvar still depends on 3.3.1 and there are a few patches that would be for OpenCV 3.3.1, but as I see it, there is no longer OpenCV3.3.1 in the source, right?

You might have missed this step on the instructions `2 - Install the ar_track_alvar package (pull request 89)`, that replaces the `libalvar`. I tested it and it ran fine. We will have to package the ar_track_alvar better somehow to merge it with the repo though.. (maybe package it into a debian and install it with the build_debians instructions)

marinagmoreira, 2023-05-31:
Closing this since we have https://github.com/nasa/astrobee/issues/662

closed 2023-05-31 by marinagmoreira