nasa/astrobee #728: Repeated haz cam extended convert scripts?

marinagmoreira, 2023-05-02:
I was cleaning up the localization_analysis and adding documentation..
It seems that the convert_depth_msg.cc (https://github.com/nasa/astrobee/blob/master/tools/localization_analysis/tools/convert_depth_msg.cc) might be repeated functionality with the script currently used by the bag_processing in the pico_driver node (https://github.com/nasa/astrobee/blob/master/hardware/pico_driver/scripts/pico_split_extended.py)

It seems that the c++ version doesn't take in a config file like the python one?
If they do the same thing, we should def delete one of them :)
Could you comment on this? @rsoussan @trey0

trey0, 2023-05-03:
If by config file for `pico_split_extended.py` you mean the `*.npy` file that needs to be calibrated for each depth camera S/N... that is needed because the script turns the depth image part of the extended message into a point cloud.

The `convert_depth_msg.cc` script is only extracting the intensity image from the extended message and ignoring the other sections. It's kind of a subset of the Python script, but might differ in the details of which other messages from the original log file are preserved, etc.

If we think it's important to consolidate these tools and also important to preserve the exact behavior of the C++ tool, I think the logical thing to do would be to modify the Python script by adding some additional arguments that allow it to replicate the C++ tool behavior.

Also seems like a low-priority task.

marinagmoreira, 2023-11-13:
closing issue since it's a low priority task that doesn't add much

closed 2023-11-13 by marinagmoreira