nasa/astrobee #738: Can't open large files on the astrobee

luisa-mao, 2023-07-12:
Attempting to open a ~2G surf map when constructing a `SparseMap` object on the astrobee robot results in a " Value too large for defined data type" error.

See this issue:
[(https://github.com/Cyan4973/xxHash/issues/451)]

Relevant code from astrobee/localization/sparse_mapping/src/sparse_map.cc is shown below:
```cpp
  int input_fd = open(protobuf_file.c_str(), O_RDONLY);
  if (input_fd < 0) {
    std::cout <<" Error code "<< strerror(errno) << std::endl;
    LOG(FATAL) << "Failed to open map file: " << protobuf_file;
  }

Pedro-Roque, 2023-07-18:
Hi @luisa-mao , 

A possible fix might be changing the data-type of input_fd to `int64`, can you check if that solves the issues and report back? Thanks in advance!

bcoltin, 2024-07-08:
Ryan lowered the number of features so we can now use the SURF maps.

closed 2024-07-08 by bcoltin