nasa/astrobee #741: On dealing with multiple robot types

Pedro-Roque, 2023-07-18:
This is more of a design kind of issue than an actual issue, but I thought this was the more appropriate way to raise it.

When supporting multiple hardware platforms using the Astrobee software stack, a few changes in the software structure need to be taken care of:
1. Most hardware-related interfaces should be moved to the hardware folder. For now, these seem to be mostly drivers, but nodes such as the `light_flow` which are Astrobee-hardware specific maybe could be moved as well
2. How to deal with different LLP and MLP interfaces? Have a separate `robot` folder for each, or have them on submodules?
3. Descriptions and media: we created different submodules/have our own `astrobee_media` . What could be the best way to standardize this? Are there ideas already or should we come up with a good way to methodically do it?

Most of these problems are being addressed in https://github.com/DISCOWER/astrobee `main` branch - which for us is a ros2 setup. Support for Ubuntu 22.04 has also been added with support for OpenCV4 (thanks @ana-GT @joris997!) and dependencies source installed into `~/.local` (as opposite to have debians installed system wide, which collided with previous system dependencies).

Maybe some time next we could have a quick call just to share some progress and updates/hurdles on this? Tagging @trey0 @marinagmoreira

Cheers!
Pedro,

marinagmoreira, 2023-07-18:
Hi @Pedro-Roque fill in your availability :) https://doodle.com/meeting/participate/id/eZ0Gvl6b

bcoltin, 2024-07-08:
We talked about this and I believe we have figured it out!

closed 2024-07-08 by bcoltin