nasa/astrobee #749: Check transforms in ROS2

marinagmoreira, 2023-10-25:
It was reported by @ana-GT that the transform between world/rviz and body is not showing up in Rviz when running:

ros2 launch astrobee sim.launch.p dds:=false rviz:=true

marinagmoreira, 2023-11-02:
Hi! I just checked this there's a couple of things you need to do to get things running.
When you compile it needs to have the --symlink-install on, as there is a weird bug that the gazebo stuff doesn't get found properly:
`colcon build --symlink-install`
then when starting the sim, use:
`ros2 launch astrobee sim.launch.py gtloc:=true debug:=true rviz:=true pose:="10.5 -9 4.5 0 0 0"`

- gtloc is needed because the localization is not converted to ros2 yet (gtloc makes the simulation use the ground truth)
- debug is optional, but the only way the gazebo error prints are displayed
- pose is optional too, but since the docking and undocking has not been tested I usually start the robot already floating
- the dds stuff is scrapped, since ros2 uses dds already

ana-GT, 2024-02-14:
Thanks! (coming back to this after a while). Running the simulation as stated above shows up the robot correctly in Rviz. 

I have a question on the current capabilities of Astrobee in ROS2. For instance, I can see that there are actions available to command the robot to move the arm and to move it itself (/mob/motion /beh/arm). When I try using these, their respective components die. For instance, when trying to send a MOVE command to /mob/motion, I get:

```[component_container_mt-22] [INFO 1707953255.046521803] [ff_action]: FFS: In waiting for response if. (CancelCallback() at /src/astrobee/install/ff_util/include/ff_util/ff_action.h:300)
[ERROR] [component_container_mt-22]: process has died [pid 145081, exit code -11, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container_mt --ros-args -r __node:=mlp_mobility -r __ns:=/'].```

Is there a list of services/actions that are expected to work? I was going to dig deeper into this error, but figured to ask first, in case these actions are not fully  supported to work in ROS2 (if they are, then I can search for the bug).

Thanks!

marinagmoreira, 2024-03-05:
> Thanks! (coming back to this after a while). Running the simulation as stated above shows up the robot correctly in Rviz.
> 
> I have a question on the current capabilities of Astrobee in ROS2. For instance, I can see that there are actions available to command the robot to move the arm and to move it itself (/mob/motion /beh/arm). When I try using these, their respective components die. For instance, when trying to send a MOVE command to /mob/motion, I get:
> 
> ```
> 
> [ERROR] [component_container_mt-22]: process has died [pid 145081, exit code -11, cmd '/opt/ros/rolling/lib/rclcpp_components/component_container_mt --ros-args -r __node:=mlp_mobility -r __ns:=/'].
> ```
> 
> Is there a list of services/actions that are expected to work? I was going to dig deeper into this error, but figured to ask first, in case these actions are not fully supported to work in ROS2 (if they are, then I can search for the bug).
> 
> Thanks!
> 
> PS.- Using the latest ubuntu-ros-rolling-20.04 docker for testing this

Hi Ana! Sorry I haven't have time to test this, I'll prob have time in the next couple of weeks. All the actions are still very experimental, I don't think anything is properly tested yet

closed 2025-01-24 by marinagmoreira