nasa/astrobee #788: ROS2 Astrobee simulation - executive conversion

ana-GT, 2024-06-05:
Hi. Recently I needed to run Astrobee in ROS2 and wanted to be able to do the following:
1. See the images from dock/nav cam.
2. Use the undock behavior.
3. Send the Astrobee simple one-point goals.

To use undock, we need the executive working. I took as reference Katie's PR: https://github.com/nasa/astrobee/pull/777 and added the following modifications:

1. Converted sys_monitor.
2. Fix the way in which ConfigClient is given a node argument (otherwise it crashes).
3. Fix a few bugs in the time utilities (seconds were being converted to nanoseconds).
4. Fix a few bugs in the launch files (specifically, use_sim_time was not being set in the right parameter field).
5. Upgrade Gazebo to be version 11.14. Current version in the rolling image is 11.11 (which makes sense, as this is the latest version that was available last April, when the rolling image for astrobee was last built). This version is needed for the nav_cam/dock_cam sensors (wideanglecamera) to be parsed.

My changes are in this branch: https://github.com/traclabs/astrobee/tree/ros2_executive .  With these changes I am currently able to undock the robot and move it around (video attached).

I'd be happy to send a PR if it is helpful, thanks!

https://github.com/nasa/astrobee/assets/555379/6a7efc22-96b6-4e38-a2fc-88773ff03eff

ana-GT, 2024-06-05:
Forgot to mention: To test:

1. Start the robot: ```ros2 launch astrobee sim.launch.py gtloc:=true debug:=true rviz:=true```
2. In another terminal, call undock: ```ros2 run executive teleop_tool -undock```
3. Once the robot is undocked, move it around: ```ros2 run executive teleop_tool -move -relative -pos "0 2 0.5"```

marinagmoreira, 2025-01-24:
Hi @ana-GT ! I can't believe I didn't see this earlier! Yes, absolutely let's get this merged into the ros2 branch!